Hypfer/Valetudo #2: Need support of "Go" and "Zoned cleanup" functions.

Metori, 2018-07-16:


maxpautsch, 2018-08-21:
is anyone working on these functions?

maxpautsch, 2018-08-22:
if anyone wants to work on this: i just created a starting point. the following commit has initial support for goTo and zoned cleanup. currently the coordinates have to be entered as to be seen in the screenshot. good for debugging. should be added to the map tab later on with area select using touch functionality. will try this soon. https://github.com/maxpautsch/Valetudo/commit/d1bedce3f2ef02139ef0ecc240375e7e74aac3ed
[image]

maxpautsch, 2018-08-22:
The latest commit on my fork adds an output of the position to the map tab. However, The go to functionality is not working with these coordinates. Perhaps a conversion is neccessary.

maxpautsch, 2018-08-24:
#15

Bennedetto, 2018-09-02:
I introduced a map drag and drop goto functionality in https://github.com/Hypfer/Valetudo/pull/19#issue-212562712. This commit also introduces conversion from robot universe to world (map/window) universe. But it modifies the map to use the generated png images as well - so I do not know if its going to be merged soon,..

[image]

At this very moment it is not able to save / modify the goto position - but is is on my todo. The same goes for zones (modify by drag / drop the edges). I suggest to apply colors for them as well - As the ability to show them on the (generated) map.

WolfspiritM, 2018-09-03:
Hi.

As I wrote in the pull request already I'm somehow missing the drag/drop and zoom functionality of the app. That's why I've been working on a map-viewer in the past days. I think keeping the drawing logic on the fronendside has many advantages. The only reason why it might lag at the moment is cause of the inefficient drawing logic.

That's what I've been working on:
https://i.gyazo.com/6a016746fb446b2e9de73ba70fe99459.mp4

I've to say that putting everything into the index.html in general is very ugly and I'd really appreciate if this project would start using some kind of frontend engine (vue.js for example) to seperate the code into different components and I'd (personally) love to see typescript being used.

The map-viewer I made is completly in typescript and independend of the index.html

The index.html contains only:
``` js
                var container = document.getElementById("current-map");

                var map = new XiaomiMap(container);
                container.addEventListener("goto", function(){
                    ...goto code
                });

     ...
                    fn.request("api/map/latest", "GET", function (err, res) {
                        loadingBar.removeAttribute("indeterminate");
                        if (!err) {
                            map.updateMapData(res);

```

Bennedetto, 2018-09-11:
> Hi.
> 
> As I wrote in the pull request already I'm somehow missing the drag/pinch and zoom functionality of the MiHome app. That's why I've been working on a map-viewer in the past days. I think keeping the drawing logic on the fronendside has many advantages. The only reason why it might lag at the moment is cause of the inefficient drawing logic.
> 
> That's what I've been working on:
> https://i.gyazo.com/6a016746fb446b2e9de73ba70fe99459.mp4
> 
> I've to say that putting everything into the index.html in general is very ugly and I'd really appreciate if this project would start using some kind of frontend engine (vue.js for example) to seperate the code into different components and I'd (personally) love to see typescript being used.
> 
> The map-viewer I made is completly in typescript and independend of the index.html
> 
> The index.html contains only:
> 
> ```js
>                 var container = document.getElementById("current-map");
> 
>                 var map = new XiaomiMap(container);
>                 container.addEventListener("goto", function(){
>                     ...goto code
>                 });
> 
>      ...
>                     fn.request("api/map/latest", "GET", function (err, res) {
>                         loadingBar.removeAttribute("indeterminate");
>                         if (!err) {
>                             map.updateMapData(res);
> ```

I would also love to have more structure in the sourcecode of valetuo. Feel free to post your changes.

WolfspiritM, 2018-09-11:
Yes. I will publish the changes as a MR. I also moved the Index.html to vue
components as typescript and fixed routing. I will try to get it done asap
:-)

WolfspiritM, 2018-11-03:
I'm sorry for the long delay but I was very busy. I now finally got some more time to work on this.
You can see the current state on https://github.com/WolfspiritM/Valetudo/tree/vue-integration

The vue integration is done and should work fine.
A few things I will take a look at over the next days:
- I noticed that Manual Control seems to respond with just hovering I think it should only work when clicked.
- The map still needs some work and I will give it more attention over the next few days. Maybe I will also add some webworker to the map view but it already seems very fast and I don't notice and client lag.
- I will also take a look at the new travis implementation cause that needs to build the frontend beforehand.
- Maybe an overall cleanup in the end...

I did a few changes to how the map is handled on the server side. It is now send in binary base64 (maybe we can even change that to binary completly) for smaller size and better handling in JS.

I also made more things configurable (SHM, LOG, ...) for developing and you can now drop a ".env" file in the root folder where your environment variables are stored. For example mine looks something like this:
``` .env
VAC_ADDRESS=192.168.178.42
VAC_TOKEN=xxxxxxxxxxxxxxxxxxxxxxxxx
VAC_SHM=./debug/shm
VAC_LOG=./debug/log
VAC_TMP_PATH=./debug/temp
VALETUDO_CONFIG=./debug/config
```

Also I fixed an issue with the remote app that tried to access the map before it has been written.

I know there are many changes and I sadly only have one commit. If you want I can try seperating the changes into multiple commits but to give you a quick overview it was faster like that.

The frontend part can be build with ```npm run build:frontend```.
For CI and building a package I also added ```npm run build``` which does run build:frontend and pkg at once.

The frontend currently uses the fetch api to call the backend.
For support of IE this might not be enough and needs a polyfill: https://caniuse.com/#search=fetch

The API calls work like that:
Component -> Vuex Store -> ServerAPI -> fetch
There is also a Mock API included which can be switched to for development (src/api/index.js at the bottom).
It fakes all calls.

Metori, 2018-11-20:
How the Zoned clean and GoTo works now? Is it using last map generated by robot for zone calculation? Or the default map? Thanks.

Hypfer, 2019-02-19:
Supported with 0.2.0

closed 2019-02-19 by Hypfer