Hypfer/Valetudo #5: Add "Mop" to the "Cleanup mode" (fan speed)

jorsmm, 2018-07-24:
Apart from "**Quiet**", "**Balanced**", "**Turbo**" and "**MAX**" there is a "**Mop**" Cleanup mode. 
/api/current_status shows `"fan_power":105` in this mode.

Could you please add this option (Mop) to the "Select fan speed" list?

Thanks!!

Hypfer, 2018-07-25:
After stopping the cleaning process and starting a new one, the robot defaults to a non-mop fanspeed.

However the robots response to `get_status` still incorrectly states `fan_power: 105` 🤔

leonardpitzu, 2018-12-03:
the fallback to the non-mop fanspeed is what happens also with the xiaomi cloud/original app. if you start a new cleaning process though it is not mopping any more so there might be a bug on ciaomi's side.

timbru31, 2019-05-05:
Is the 

> After stopping the cleaning process and starting a new one, the robot defaults to a non-mop fanspeed.
However the robots response to get_status still incorrectly states fan_power: 105 🤔

The blocking issue here? Since the 105 has some effect I always have to manually toggle the mop mode.

timbru31, 2019-05-10:
Added via #213  - can you close this issue @Hypfer ?

closed 2019-05-10 by Hypfer