Hypfer/Valetudo #6: Home-menu not working (but Map is)

aebgit, 2018-07-27:
1st: great project. thanks for sharing!

I Installed latest beta-release v0.0.3 from the release-section (pre-built) on my Gen2 and followed the instructions from `deployment/Readme.md`
The server starts and I can access it via web-browser, but I cannot use any of the functions on the home-screen. Everything is greyed out. The map and also the settings-screen are working. I assume there is something wrong with the local miio-interconnect between rockrobo <-> valetedo on the vac. I get plenty of error-messages in `/mnt/data/rockrobo/rrlog/miio.log `

```
0x21 0x31 0x00 0x60 0x00 0x00 0x00 0x00 0x04 0x70 0xd2 0x17 0x00 0x00 0x02 0x7a 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[20180727 16:55:49] [INFO] OT protocol diagrams come...
[20180727 16:55:49] [WARNING] OT MD5 not match.

0x21 0x31 0x00 0x60 0x00 0x00 0x00 0x00 0x04 0x70 0xd2 0x17 0x00 0x00 0x02 0x7a 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[20180727 16:55:49] [INFO] OT protocol diagrams come...
[20180727 16:55:49] [WARNING] OT MD5 not match.

0x21 0x31 0x00 0x60 0x00 0x00 0x00 0x00 0x04 0x70 0xd2 0x17 0x00 0x00 0x02 0x7a 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[20180727 16:55:49] [INFO] OT protocol diagrams come...
[20180727 16:55:49] [WARNING] OT MD5 not match.

0x21 0x31 0x00 0x60 0x00 0x00 0x00 0x00 0x04 0x70 0xd2 0x17 0x00 0x00 0x02 0x7a 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[20180727 16:55:49] [INFO] OT protocol diagrams come...
[20180727 16:55:49] [WARNING] OT MD5 not match.
```

Hypfer, 2018-07-27:
Since the buttons only get enabled as soon as the status is known I suppose that the communication doesn't work at all.

But why? Did you change something in the vacuums firmware? Which FW version are you running?

aebgit, 2018-07-27:
The problem seems to be related to the redirected hostnames in `/etc/hosts`. After I added the german Xiaomi-servers and rebooted again it was working (see last two lines).

```
root@rockrobo:~# cat /etc/hosts
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

10.5.5.5      ott.io.mi.com
10.5.5.5      ot.io.mi.com

10.5.5.5      de.ot.io.mi.com
10.5.5.5      de.ot.io.mi.com
```

closed 2018-07-27 by aebgit