Hypfer/Valetudo #7: FeatureRequest: homebridge-module that makes use of the valetudo-api

aebgit, 2018-07-28:
Dear all,

I'm not sure if this is the right place for such a discussion, so please excuse me if I'm wrong here.
Unfortunately I'm very bad at programming, so I integrated Valetudo into my homebridge (Homekit) environment in a very bad manner. (but it works ;-) 
I use the homebridge-plugin `cmdAccessory` to fire curl-commands with the API-calls for Valetudo. in my homebridge-config this looks similar to this:

```
 {
"name": "RockroboDock",
"on_cmd": "curl 'http://<rockrobo-ip>/api/drive_home' -XPUT -H 'Origin: http://<rockrobo-ip>' -H 'Host: <rockrobo-ip>' -H 'DNT: 1' -H 'Connection: keep-alive' -H 'Content-Length: 0' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: de-de' -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_5) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/11.1.1 Safari/605.1.15' -H 'Accept: */*' -H 'Referer: http://<rockrobo-ip>/'",
"off_cmd": "curl 'http://<rockrobo-ip>/api/pause_cleaning' -XPUT -H 'Origin: http://<rockrobo-ip>' -H 'Host: <rockrobo-ip>' -H 'DNT: 1' -H 'Connection: keep-alive' -H 'Content-Length: 0' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: de-de' -H 'User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_5) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/11.1.1 Safari/605.1.15' -H 'Accept: */*' -H 'Referer: <rockrobo-ip>/'",
"polling": true,
"interval": "10",
"state_cmd": "/bin/false",
"manufacturer": "aebgit",
"type": "Switch"
},
```

I wonder if somebody more skilled than me, can use this as inspiration for a homebridge-module that uses the Valetudo web-api. In my eyes this would be a great benefit to the homebridge- and  rockrobo-community, because the current solutions (like `homebridge-xiaomi-mi-robot-vacuum`) that make use of the xiaomi-token are very unreliable.

Bet regards
Andreas

**references:**
https://github.com/Hypfer/Valetudo
https://github.com/nfarina/homebridge
https://www.npmjs.com/package/homebridge-xiaomi-mi-robot-vacuum
https://github.com/YinHangCode/homebridge-mi-robot_vacuum

leonardpitzu, 2018-12-27:
Try this: https://github.com/nicoh88/homebridge-xiaomi-roborock-vacuum

Hypfer, 2019-04-06:
Replaced with the mqtt interface in 0.2.0

closed 2019-04-06 by Hypfer