Hypfer/Valetudo #17: Add bin time / timestamp to data

markus0m, 2018-08-30:
Bin time is recorded by the robot in seconds.
File is /mnt/data/RoboController.cfg, value is bin_in_time:

`RoboController : 
{
  runtime : 
  {
    bin_in_time = 265;
    time_slept = 0;
    sound_volume = 70;
    clean_id = 45;
    carpet_mode_enabled = 1;
    carpet_mode_curr_integral = 450;
    carpet_mode_curr_highwater = 500;
    carpet_mode_curr_lowwater = 400;
    carpet_mode_stall_time = 10;
    fan_power = 60;
    temp_fan_power = -1;
  };
};
`

While this doesn't let us react to removing the bin or see whether it's in, Valetudo can display the cleaning time since last bin removal.

Keeping and displaying an additional timestamp every time this value jumps to 0 or decreases would also be great.

markus0m, 2018-08-30:
There's also an event in MiIO log:
[20180830 20:57:35] [DEBUG] ot_agent_recv_handler_one(): fd:15, msg:{"id":491,"result":[{"msg_ver":8,"msg_seq":16,"state":3,"battery":99,"clean_time":63,"clean_area":1862500,"error_code":0,"map_present":1,"in_cleaning":0,"fan_power":75,"dnd_enabled":1,**"events":["bin_full"]**}]} length:208 bytes
Seconds threshold for this is unknown

Hypfer, 2018-09-17:
Okay so I just got the message. Value was 47979, however I didn't check it for like a week so its most likely less than that.

Hypfer, 2019-06-13:
To find out the bin_time value which triggers this, we're now able to (hackishly) take this https://github.com/Hypfer/Valetudo/blob/master/lib/miio/Dummycloud.js#L154 and add a readFileSync followed by a console.log for the RoboController.cfg.

Then we just wait

Geertvdc, 2019-09-08:
This feature would be awesome! I want to program My vacuum to go to the trash can after cleaning. As soon as i read an emptied dustbin and the vacuum is not docked i send him home

robotastic-io, 2020-05-07:
Any update on this? I would love to have an automation where the vacuum waits near the dust bin after a cleaning run when x days of last bin removal have passed. 

Having a timestamp of last bin removal would be great for sufficient for this.

rumpeltux, 2020-05-08:
Contributions are welcome :-)

closed 2021-01-09 by Hypfer

theorician, 2021-01-09:
@Hypfer Did this get closed because there was a PR with this feature added, or because you're cleaning house of old issues?