Hypfer/Valetudo #26: Start Manual Control immediately terminates on first start

Yky, 2018-09-23:
Precondition: 
The v2 robot is in the dock

Problem description:
If I start manual control, the button stop manual control becomes active only for a split second, then the start manual control button becomes active again. 

The robot says "using remote controls" followed by "finished using remote controls" right afterwards. 

If I repeat this process a few times at some point the robot starts the fans and does not "finish using remote controls" immediately, such that manual control is possible. 

Expected behavior:
Keep manual controls running after starting them until "stop manual control" is clicked

FelixUster, 2018-09-24:
I can confirm this bug for Gen 1

Hypfer, 2019-04-28:
#192

closed 2019-04-28 by Hypfer