Hypfer/Valetudo #27: Show firmware version in settings

Yky, 2018-09-23:
It would be useful to check the firmware version in the Settings, as is possible in the Xiaomi app.

TobiasSchaffner, 2018-09-25:
Maybe somebody can help me out here. I have no to little experience with webdevelopment.
I think this can be done with a "miIO.info" query. (I assume this in reference to https://github.com/ioBroker/ioBroker.mihome-vacuum/blob/master/lib/comands.js) Is this correct?
I created a patch that adds the information to the settings page but I get a connection error and have a hard time tracing this down.
https://github.com/TobiasSchaffner/Valetudo/commit/a57d83dbf60d956f42b34e6c4d1ce0b005d8246b

Hypfer, 2018-09-26:
miIO.info only works if the robot is provisioned and has an active cloud connection.
Hence it should not be used for anything in valetudo. 😔

TobiasSchaffner, 2018-09-26:
...Thanks anyway. 👍 Can you maybe point me to some resources where I could have found that information or did you figure that out by your self?

Hypfer, 2018-09-26:
I just figured it out by myself. 
There was also some validation happening in the telegram dustcloud group:

> Hypfer, [23.06.18 13:39]
> if yes, I'd assume that miIO.info is not available on devices which aren't connected to the cloud

>Teemu R, [23.06.18 17:23]
>the vacuum is not responsive to infos when it has no cloud connectivity, that's why python-miio avoids using it :P

rytilahti, 2018-09-26:
Yeah, it's an unfortunate implementation bug, some miIO devices respond with broken data (like the extension coord) when no network access is available. Anyway, looks like the robot has the version information available here:
```
# cat /etc/os-release
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.3 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
ROCKROBO_VERSION=3.3.9_003194
```

stefanct, 2018-10-20:
This works fine since 0.0.7 thanks to https://github.com/Hypfer/Valetudo/commit/33b2d844a5bd4746140cc56e54553508be9f2e4e.

closed 2018-10-20 by Hypfer