Hypfer/Valetudo #32: Check for firmware update

chbndrhnns, 2018-10-06:
I would love to see a notification if the currently used firmware is out-of-date. This way I only need the Mi app to update the firmware (as I assume this might be difficult to achieve using Valetudo?).

Hypfer, 2018-10-06:
Since new firmware updates will remove valetudo as well as possibly lock the robots firmware I don't think that this is a good idea.

chbndrhnns, 2018-10-06:
Ok, I was not aware. This would mean updating is generally discouraged until it is clear that the firmware does not break anything...

Hypfer, 2018-10-06:
Yes.

Afaik, the latest firmware tries to prevent downgrades. 🤔 Not 100% sure if it locks you out yet, though

akeilox, 2018-10-06:
on this topic how do you check new firmware for locking? Is there a particular source you follow for such notices?

Hypfer, 2018-10-06:
Dennis Giese mentioned something about that in the official dustcloud telegram group.

akeilox, 2018-10-06:
Thanks for your reply. I didnt check dustcloud telegram in ages (muted), guess its about time to catch up if xiaomi introducing locking

closed 2018-11-05 by Hypfer