Hypfer/Valetudo #33: WebGui keeps showing Loading State again and again

BLAUSICHT, 2018-10-18:
Hello,

first of all great Project :)!!!!

I have a weird Problem.
I open the WebGui and it shows up correctly. But than a few seconds later, randomly it starts loading and loading and cant get informations from the Vaccum. (Loading state...) for maybe 5 Minutes and than its okay again and than 5 Minutes later it starts againg loading and loading.

Can someone help me, fix this issue.
I have a rooted firmware (rooted with Dustcloud)
FW Version: 3.3.9
FW Build: 001518

How did i install Valetudo: i copied the latest final Binary 0.0.7 with the valetudo.conf.
If you need more Informations, tell me :)

Kind regards,

BLAUSICHT

trohnjavolta, 2018-10-20:
Same for me. Before blocking xiaomi servers in /etc/hosts everything worked.
Since the description states: ...requires no cloud connection whatsoever...
I assumed one can use it without a connection to xiaomi servers... am I wrong?

My /etc/hosts looks like this:
```
127.0.0.1   localhost
127.0.1.1   rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1       ip6-allnodes
ff02::2       ip6-allrouters

0.0.0.0      awsbj0-files.fds.api.xiaomi.com
0.0.0.0      awsbj0.fds.api.xiaomi.com
0.0.0.0      ott.io.mi.com
0.0.0.0      ot.io.mi.com
```

Hypfer, 2018-10-20:
https://github.com/Hypfer/Valetudo/blob/master/deployment/Readme.md

BLAUSICHT, 2018-10-21:
Sorry, my IPTABLES looks like this 

#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
iptables -A OUTPUT -d 10.5.5.5 -j REJECT
iptables -A OUTPUT -d 10.0.0/24 -j ACCEPT
iptables -A OUTPUT -d 127.0.0.0/8 -j ACCEPT
iptables -A OUTPUT -p udp --dport 123 -j ACCEPT
iptables -A INPUT -p udp --sport 123 -j ACCEPT
iptables -A OUTPUT -j DROP
exit 0

And My /etc/hosts looks like this

127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters
#10.5.5.5       de.ott.io.mi.com
#10.5.5.5       de.ot.io.mi.com
10.5.5.5      ott.io.mi.com
10.5.5.5      ot.io.mi.com

But it does keep have problems every 5 Minutes.

Hypfer, 2018-10-21:
I guess you could try replacing 10.5.5.5 with something in the 192.168.x.x range if your home network is 10.x.x.x

BLAUSICHT, 2018-10-21:
i tried this right now. And the Same issue occurs.. :'(

127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters
#10.5.5.5       de.ott.io.mi.com
#10.5.5.5       de.ot.io.mi.com
192.168.0.99      ott.io.mi.com
192.168.0.99      ot.io.mi.com

iptables -A OUTPUT -d 192.168.0.99 -j REJECT
iptables -A OUTPUT -d 10.0.0/24 -j ACCEPT
iptables -A OUTPUT -d 127.0.0.0/8 -j ACCEPT
iptables -A OUTPUT -p udp --dport 123 -j ACCEPT
iptables -A INPUT -p udp --sport 123 -j ACCEPT
iptables -A OUTPUT -j DROP

BLAUSICHT, 2018-10-23:
has someone an idea?

BLAUSICHT, 2018-12-28:
i did a Factory Reset on my Roborock and installed everything new.
Now the Problems are Solved :).

closed 2018-12-28 by BLAUSICHT

theFork, 2019-04-03:
I got the same problem. It's weird, that the map keeps updating, but the home tab keeps loading the state. I think we should reopen this one.

Hypfer, 2019-04-03:
Currently, the map is read from fs. Thats why its updating while everything which uses the miio interface doesn't respond.

Endless loading times are often a sign of out-of-sync messageIds which can happen due to misconfigurations or valetudo crashing.

VirtualGit, 2019-04-08:
I got the same issue !
Web gui is working well, but all buttons remain disabled.
* api/get_config returns normal json
* api/current_status freezes and never returns data

My vacuum can't access Internet (blocked by my router). If I allow the vacuum to access Internet, I get status and gui is refreshed only once (until I reboot the vacuum).

valetudo.log seems ok, but prints two lines about the token, make me feels that's when I got my unique status :
```
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Got token from handshake: 5a597a4d43534630685048414e365375
Got an expired token. Changing to new
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
```
Firmware : v11_001702

Any idea ? Note that I got the same issue with mirobo (python-miio tool)

cryptomilk, 2019-04-08:
Did you follow https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions

VirtualGit, 2019-04-08:
Yes, the only difference I see is the firmware version (and that I used a non fullos pkg version). My deep feeling is the robot no longers answer requests when isolated from Internet. Before rooting it, I've quickly connected it to the official App to check the robot is ok. Could this be related ?

cryptomilk, 2019-04-08:
Did you use the latest version of the imagebuilder.sh script?

Hypfer, 2019-04-08:
You can't use the Xiaomi cloud with Valetudo.

fvollmer, 2019-04-08:
I had the same problem. Then tried a reboot of the robot and now the gui works.

VirtualGit, 2019-04-08:
New tries. I had an imageBuilder.sh of March 20th. I updated it and try again.
It seems to be equal first, then I decided to play with device.token file. Indeed, I've seen that the token displayed in valetudo was different than the one in device.token.
I can't remember exactly what I did, but it's something like :
* Inject token given by valetudo into device.token (If I remember : echo xxxxxxx | xxd -p -r > device.token) : without success (and maybe no more gui at all)
* make device.token empty : without success (and maybe no more gui at all)
* Restore the backed-up device.token file : vateluto gui works well !
Hard to understand what was happening but for sure it's not related to Valetudo but rather to some kind of protection of the robot software.
Thanks for help.

Buffi66, 2020-02-23:
In my case I could "fix" the problem with stucking GUI, resp. text "Loading state..." and no button works by using a faster machine for the GUI. On my old Android tablet, about 5 years old, I had the problem but on my relatively new Samsung S2 tablet everthing works like a charme.

Hypfer, 2020-02-23:
@Buffi66 The browser of the 5 year old tablet probably doesn't support newer web features