Hypfer/Valetudo #34: High resources usage of map view

stefanct, 2018-10-20:
The way the map tab is implemented seems to require quite some browser resources. I have not tested in detail but the desktop versions of Firefox and Chromium alike get noticeably slaggier when the map is opened. GPU memory in chromium's process memory peaks at about 300MB. The resident size of the respective Firefox tab up to about 2GB(!).
I detest web development, javascript and ruby so I'am a bit out my depth here but I guess things would improve if the canvas (or its contents) are not recreated on every update(?) and left to the GC to clean up but either simply updated or manually destructed.

Hypfer, 2019-02-19:
Should be fixed in 0.2.0

Please confirm

stefanct, 2019-02-23:
The situation has at least very much improved since my report so that I would deem it OK now, thanks.

closed 2019-02-23 by stefanct