Hypfer/Valetudo #35: "add area to existing zone" does not work

sebdoan, 2018-10-20:
Not sure if I'm doing something wrong but I have this issue. 

**Steps to recreate:** 
1. add the X1, Y1, X2, Y2 coordinates to a new zone and call it "Hallway"
2. add the X3, Y3, X4, Y4 coordinates to another zone and choose "add area to existing zone" and select "Hallway"
3. Select the Hallway as a zone to start cleaning

**Result:**
The MiHome app shows a rectangle to indicate that only X1, Y1, X2, Y2 coordinates are being cleaned.

**Expected Result**
The MiHome app shows a 2 rectangles to show that X1, Y1, X2, Y2 and X3, Y3, X4, Y4 coordinates are being cleaned.

P.S. Also as a feature request it would be nice to see positive confirmation of the X, Y values being used for each zone.

sebdoan, 2018-10-25:
I'll test this in the coming day as I've had to upgrade the firmware of the MiRobot and lost my root access. 

On a side note, why have you elected to use a different coordinate system compared to the typical 25500 values? https://community.home-assistant.io/t/howto-xiaomi-vacuum-zoned-cleaning/51293

bendews, 2018-10-25:
@sebdoan the coordinate system is the same just with the values (the charging base) starting at 0. 
The coordinates are in millimetres, so having the base station be 0 makes it easier to visualise going *n* meters forwards and *n* metres to the left without having to do heaps of math 😄

```
valetudo_X = 10000
valetudo_Y = 12000

normal_X = 25500 - 10000
normal_Y = 25500 - 12000

> normal_X: 15000
> normal_Y: 13500
```

PBue, 2018-12-23:
Hi,

I have the same problem (Release **0.0.9** with **Gen 1** Vacuum).

**What I also tried:**
I manually extended the array in "/mnt/data/valetudo/config.json" like this:
`{"spots":[],"areas":[["My Room",[[2766,9332,4653,7578,1],[3048,5835,264,10335,1]]]]}`

**The result:**
The vacuum still only cleans the first rectangular; the second one is ignored.
And if I add another zone "My Room2" the second rectangular of "My Room" is removed again.

Hope this helps.

Happy Holidays :-)

PR #65 (2018-12-23): Fixes problem while adding a zone to an existing area (
Fixes #35
closed 2018-12-23 by Hypfer