Hypfer/Valetudo #37: Use Map to create zones and pinpoint for GoTo

bendews, 2018-10-21:
Super obvious and I'm sure its on everyone's wishlist but hadn't seen it documented yet.

It would be great to mirror some of the functionality of the official app and allow the user to both pinpoint areas for GoTo and click+drag to create zones. 👍 
I have had a look at the front-end JS and the data returned from the API and can see it's not the easiest, but hopefully is possible!

stefanct, 2018-11-19:
It would already help tremendously if the current (map/real-world) coordinates of the mouse pointer would be displayed in the map view. This should be way easier to implement than the full customization of zones and gotos within the map view.

cryptomilk, 2018-12-21:
A zone feature has been added if if you build from source install the new version and go to: http://rockrobo/zone

stefanct, 2019-01-05:
Works well, nice, thank you! Still a bit hidden though ;) Some issues I noticed: resizing the (browser) window wreaks havoc; it would be nice to be able to edit the coordinates. Most intuitively would be of course to drag'n'drop the edges or corners, but I think making the text fields writable and redrawing the red border on new inputs would be good enough.

matthiasharrer, 2019-01-05:
Hi,

after using the Mi Home app for a while, I am back working on the zone selection stuff here and then (Having fresh ideas :).

I would like to have the zone selection map somehow integrated with the rest of the frontend (If anyone is more fluent in OnsenUI that might help). 
Also I think using the map directly for showing and changing the saved zones would be nice!

Here is what it looks like right now (https://github.com/matthiasharrer/Valetudo/tree/map-rework):
Window Resize is still not handled and the displayed coordinates are not updated when moving/resizing the map.

[image]

Hypfer, 2019-02-19:
Fixed with 0.2.0

stefanct, 2019-02-19:
Including the resize problem?

matthiasharrer, 2019-02-19:
nope .. it did strange things i had not have the time to investigate further unfortunately .. do you resize frequently?

stefanct, 2019-02-19:
No, it's not a big issue of course but I was asking because @Hypfer closed the bug and I think we should still track it in case anybody looks for something small-ish to do ;)

Hypfer, 2019-02-19:
True

closed 2019-02-20 by Hypfer

stefanct, 2019-02-23:
The GUI selection stuff is working fine (although tool tips for the buttons on the map and a distinct color for the goto point would be nice). However, I could not find a way to use the features drawn either on `/zone` or the normal map to form persistent zones/gotos. So this is actually a regression in terms of usability to set up permanent zones because one can no longer retrieve the coordinates either. Great work Matthias though, thank you!

Using the drawing method in the ordinary zone and goto setups is probably the goal isn't it? In that case - unless this is tracked in some other issue - I would not deem this as fixed.

PR #101 (2019-02-20): handles window.resize event and appropriately resizes the map canvas
This fixes the distorted map when resizing the browser window and eliminates the scrollbars present when resizing.

Closes #37