Hypfer/Valetudo #39: Over-polling map

dugite-code, 2018-10-30:
I discovered a small issue when using Home-Assistant to Poll the Map too often (every 10 seconds to get a sort of Live map). The `/tmp` directory filled and the vacuum locked up. A re-boot fixes the issue obviously.

I had done other things on the vacuum in the past so I might have had less space available than normal.

A quick work-around was to throw this into cron
`*/1 * * * * cd /tmp/maps && ls -t /tmp/maps | tail -n +2 | xargs rm --`

dugite-code, 2018-11-05:
Actually after further looking into the matter it looks like it might be a low memory issue rather than the `/tmp` partition filling up. I'v added a swap file on `/mnt/data` to see if that solves my issues

As you can see Valetudo is taking up 76% of the memory:
[image]

Hypfer, 2018-11-05:
After looking at the [code ](https://github.com/Hypfer/Valetudo/blob/master/webserver/WebServer.js#L403) for 20 minutes, I still have no idea what could be responsible for the memory leak.

However, I've noticed that on every request the robot + charger images are read from flash for no reason so I guess the whole part will be rewritten at some point.
Could you provide a specification for this interface? What endpoints are there? What does Home-Assistant expect? 
I'm not using those home automation frameworks nor did I write said code so I don't have the slightest Idea. 🙃

dugite-code, 2018-11-06:
Essentially all HA is doing is calling `YOUR.VACUUM.ROBOT.IP/api/remote/map` and then pulling the image using the mapsrc from the json response using a GET request (using the python requests library).

The only real difference between HA and using curl is the speed of the requests, normally HA will do this every 30 seconds I tuned it up to every 10 seconds, I wouldn't think that would cause an issue.

setting  drawRobot and drawCharger to false by calling `YOUR.VACUUM.ROBOT.IP/api/remote/map?drawRobot=false&drawCharger=false&scale=5&border=3&doCropping=true&drawPath=true` looks like it might fix the issue. When drawing the robot and charger at 3min run time it was rising above 25% memory, when not drawing the robot and charger it's bouncing between 16% and 20% at 5min run time.

I'll run this for a while without restarting and let you know if the memory usage stays at ~20% mark

As a side note: Having a further look into the HA settings I've enabled `limit_refetch_to_url_change: True`, This should reduce the fetching of the map image a little.

dugite-code, 2018-11-12:
After further looking into it, it's strange I am seeing the issue after all the upstart script has an oom score of 1000 so it should in theory handle itself. I am using an older firmware `v11_003194` so perhaps that is the issue.

Hypfer, 2018-12-23:
Is this still happening? Did you try remote debugging and taking a heap dump?

dugite-code, 2018-12-25:
I am still seeing the issue. I'll try to look into the heap dump soon.

I'll also look into updating the firmware as I'm using an older one

axel-kah, 2018-12-29:
I just rooted my rockrobo v1 with latest firmwarebuilder (FW v11_003194, --disable-xiaomi), dummycloud_0.1 and Valetudo 0.9 via rrcc. I really like the newly gained power over my vacuum but observed the same behaviour as @dugite-code. Basically every map related call increases the memory footprint of Valetudo process.

I did some measurements in a notebook to stress Valetudo by doing the same api call 1000 times in a row. requests was used to interact with Valetudo API and output of ps via paramiko for the process info. The memory footprint steadily increases to up to 60% before the process terminates when triggering a new map with ``api/remote/map`` and requesting the png.
[image]

Simply fetching the same png for 1000 times also steadily increases footprint but only by 0.5% points.
[image]

Doing the same for ``api/map/latest`` does also increase footprint somewhat.
[image]

I don't have any experience with node.js apps but I'll try to use the techniques described here to get a clue why the garbage collection fails: https://www.nearform.com/blog/how-to-self-detect-a-memory-leak-in-node/

Hypfer, 2018-12-30:
Thanks for looking into that and providing this graphs @axel-kah!
Sadly, I can't help much with that. I'd still love to just drop local map generation altogether which should definitely fix this issue as well :^)

This depends on #66

Hypfer, 2019-03-23:
I guess this should be fixed with https://github.com/Hypfer/Valetudo/releases/tag/0.2.2

Hypfer, 2019-03-23:
Nope. Still an issue.

Although it seems like Valetudo is not the only software experiencing it
https://github.com/oliver-moran/jimp/issues/153

Hypfer, 2019-03-23:
Doesn't happen with Node 8 it seems.

Hypfer, 2019-03-23:
Testing Node 11.12 shows no signs of the leak so I guess https://github.com/nodejs/node/issues/23862 this is the culprit.

Fixed in Node 11.10
https://github.com/nodejs/node/pull/25993

closed 2019-03-23 by Hypfer

dugite-code, 2019-03-25:
Awesome, with the map via MQTT it looks like it's now working as expected.

lance36, 2019-03-25:
Great work!