Hypfer/Valetudo #40: Error 500 when using map function on Gen 1 firmware v11_003514

ovaltineo, 2018-11-03:
Valetudo worked properly on older firmware but now returns error 500 when using map function on Gen 1 firmware v11_003514 (downloaded via FloleVac app).

Crazyachmed, 2018-11-04:
Works fine for me on 3514. I had this issue only before I let the robot run for the first time.

ovaltineo, 2018-11-04:
It definitely has a map of the house - I can see the map in the Xiaomi app and FloleVac app.  It was giving the error 500 when I tried it while charging.  I tried now while it is doing a zoned cleaning and it worked.

ovaltineo, 2018-11-04:
Sure it enough, it returns error 500 when it went back to dock to charge.  Is this a known bug?

Crazyachmed, 2018-11-05:
Now that you are saying it, it displays an old map as soon as it hits the dock

Mdleal, 2018-11-30:
I am seeing the issue with a Gen 2 running v11_001633

herrwusel, 2018-12-01:
@Mdleal same here: Rockrobo S50 with 1633 firmware.

moskovskiy82, 2018-12-02:
Same with me 1780

JohnRev, 2018-12-03:
I have the same issue with 1644 and above.
The issue does not occur on 1518.

I have noticed that in the newer firmwares, the map file in `/mnt/data/rockrobo/rrlog/` gets renamed from `navmap344327.ppm.0000.gz` to `navmap344327.ppm.0000.rr.gz`, and so the regex on https://github.com/Hypfer/Valetudo/blob/2b38868bf5df113e6fe66788502c9822ef68b71c/webserver/WebServer.js#L925 and https://github.com/Hypfer/Valetudo/blob/2b38868bf5df113e6fe66788502c9822ef68b71c/webserver/WebServer.js#L928 will fail.

I tried modifying the regex to something like:
`/^navmap([0-9]+)\.ppm\.([0-9]{4})(\.rr)?\.gz$`.
With this updated line, the navmap file is correctly identified and located, but the decryption fails. I am not sure how those `.rr.gz` files are encrypted.

Hypfer, 2018-12-03:
Firmware 1518 is the last firmware that uses the old map format IIRC.
It could be possible that the .rr.gz navmap is in a completely different format which is not (yet) supported.

A sample .rr.gz Mapfile should be interesting. If you feel like sharing your map please attach it to #44

leonardpitzu, 2018-12-03:
i'm on 1633 - what do you need, the nvmap archive or something more?

Hypfer, 2018-12-03:
Looks like they indeed changed the encryption key. I have no idea how the previous one was found so unless someone finds out more I can't do much about it.

leonardpitzu, 2018-12-03:
Can we downgrade? I think it’s the best option now.

herrwusel, 2018-12-04:
@JohnRev Do you have some hints on how to downgrade? Just do the same steps with older firmware?

JohnRev, 2018-12-04:
@herrwusel : I downgraded by flashing 1518 using flasher.py and the instructions here https://github.com/dgiese/dustcloud/wiki/VacuumRobots-manual-update-root-Howto

@leonardpitzu, @Hypfer : Another idea. How about we just use the latest map from `robot.db` with an extractor similar to https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum.gen1/mapextractor/extractor.py . This will only be needed when charging, since live maps were working fine on the newer versions, if I recall correctly.

Crazyachmed, 2018-12-04:
> This will only be needed when charging, since live maps were working fine on the newer versions, if I recall correctly.

Somewhat. The live map is currently mirrored (the saved map is not), at least in my case, and also the live map looses path data after some time, so it looks like the robot started somewhere mid cleaning.

herrwusel, 2018-12-04:
With newer firmware I had the issue that the rooms were stacking over each other. I put the vacuum into a new room and it kept showing the map from the room before as well. Have to test it with 1518 though.

leonardpitzu, 2018-12-04:
i downgraded myself this morning. i will try to get a look into how this thing in built but my free time is mostly during flights so my progress will be...

JohnRev, 2018-12-12:
There seems to be some issue with how the subfolders under `.mnt/data/rockrobo/rrlog/` are scanned.
If there is a new folder with no usable map in it, Valetudo throws a 500 error.

To test, go to `/mnt/data/rockrobo/rrlog/`, do an `ls` and see the name of the newest folder. Let's assume it's `000105.[redacted]`. Create an empty directory with a larger number `mkdir 000106.[redcated]`. Valetudo throws error 500.

The expected behavior is that Valetudo should ignore the empty folder since there was no usable map in there.

I suspect this is because of this line https://github.com/Hypfer/Valetudo/blob/529070b2fe261c738c25f0b382a81961cd3df37c/webserver/WebServer.js#L934

cryptomilk, 2018-12-20:
@JohnRev I've just build the latest valentuo with your fix but I'm still getting the Error 500 (running Gen2 v001780). I guess I will install strace to see what is going on, I think valetudo has no debug logging right?

cryptomilk, 2018-12-20:
I see that it walks all  `/mnt/data/rockrobo/rrlog/000*` directories and lists there contents, but it doesn't find what it is looking for and returns: `Error: No usable map data found`

cryptomilk, 2018-12-20:
I've changed the regex to also include rr files, but the map function still works, however I can open the file locally ...

cryptomilk, 2018-12-20:
Looking at strace I see that it reads both files the log and the map. The unzipped filesize is also correct.So it looks like the data is correctly delivered to the browser, but the browser is not able to display it for some reason ..

JohnRev, 2018-12-20:
@cryptomilk: If you can open the file, then the rrlogd patch is working. 
Maybe the issue is what I fixed in this pull request: https://github.com/Hypfer/Valetudo/pull/59 ?
Otherwise, check what is the response of http://[vauum_ip]/api/map/latest

cryptomilk, 2018-12-20:
@JohnRev I already have that fix and also applied the regex fix you posted earlier. As I said, the files are correctly read, but then the webbrowser runs havoc that firefox wants to stop the script. Acorrding to strace the file are correctly read and parsed and delivered to the browser.

JohnRev, 2018-12-20:
It would be interesting to debug, and possibly open a separate issue for this bug. Do you mind sharing your map and log files so we can try to replicate?

cryptomilk, 2018-12-21:
@JohnRev Contact me via mail and I can send you the files.

Fietspomp86, 2019-01-04:
Isn't this issue already solved from the other issue? #44 
With the patched rrlogd for me it's working when charging (Gen1, 3514 firmware)

SoulSlayerPT, 2019-01-13:
Could make the rrlogd patch to work with 1632 and 1702. Can someone provide me a link to 1780 firmware? Cannot find it anywhere..

cryptomilk, 2019-01-13:
`./imagebuilder.sh --patch-rrlogd`

https://hackmd.io/s/HkwF33oWE

SoulSlayerPT, 2019-01-13:
Tried with your link and firmware 1702. It doesn't work.. Still error 500 while charging.

sudo ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh \
>   --firmware=../firmware/v11_001702.pkg \
>   --soundfile=../firmware/english.pkg \
>   --public-key=$HOME/.ssh/id_ed25519.pub \
>   --valetudo-path=../valetudo \
>   --disable-firmware-updates \
>   --patch-rrlogd \
>   --replace-adbd
/home/pedro/rockrobo/image/imagebuilder.sh
Compatible readlink found!
Script path: ../dustcloud/devices/xiaomi.vacuum/firmwarebuilder
Generate SSH Host Keys if necessary
Decrypt soundfile ..
Unpack soundfile ..
~/rockrobo/image/fw.jgVkds/sounds ~/rockrobo/image
~/rockrobo/image
Decrypt firmware
Unpack firmware
~/rockrobo/image/fw.jgVkds/fw ~/rockrobo/image
~/rockrobo/image
Replace ssh host keys
Disable SSH firewall rule
Add SSH authorized_keys
reconfiguring network traffic to xiaomi
replacing adbd
Creating backup of rrlogd
Trying to patch rrlogd
Successfully patched rrlogd
Installing valetudo
Pack new firmware
~/rockrobo/image/fw.jgVkds/fw ~/rockrobo/image
Encrypt firmware
~/rockrobo/image
Copy firmware to output/v11_001702.pkg and creating checksums
Cleaning up
FINISHED

hagenuck1, 2019-01-13:
I just upgraded to 1702 using imagebuilder, too. But for me Valetudo & dummycloud wasn't included. Did this work for you? Maybe there is an issue with the imagebuilder currently?

SoulSlayerPT, 2019-01-13:
I did the run and i still have the error. That is why i would like to have a 1780 firmware pkg to download. to try with it and if doens't work copy the file manually.

Mdleal, 2019-01-14:
I am having the same issue with 1780 as well. Is there a way to verify rrlogd is patched?

Mdleal, 2019-01-14:
I copied the patched rrlog file from [this ](https://github.com/Hypfer/Valetudo/issues/44) issue and it's working now.

SoulSlayerPT, 2019-01-14:
Can someone upload to wetransfer or some other website the 1780 pkg so i can test also? Thanks

cryptomilk, 2019-01-14:
@Mdleal could you try if rrlogd works if you patch it using: https://github.com/JohnRev/rrlogd-patcher

SoulSlayerPT, 2019-01-14:
@cryptomilk With your last link it is working now! I patched the rrlogd and it is now working with 1702

cryptomilk, 2019-01-14:
I will change the imagebuilder script to use the rrlogd-patcher.

Maja61, 2019-01-16:
I test the patcher with an rockrobo.vacuum.v1 v3.3.9_003452, but it doesn´t help. I have still error 500 during charging.

cryptomilk, 2019-01-16:
@Maja61 Yes, as the README.md states clearly: Gen1 Firmwares >= 3514

ideasman69, 2019-01-19:
i followed this guide to the letter using v3514 on a gen1 and still get error 500 when charging. No errors from firmwarebuilder and patching of rrlog says it was successful.

cryptomilk, 2019-01-20:
Fixes: https://github.com/dgiese/dustcloud/pull/181

Maja61, 2019-01-20:
> 
> 
> @Maja61 Yes, as the README.md states clearly: Gen1 Firmwares >= 3514

Update the Robo to 3600 and again follow the guide of the patcher, but again still error 500.

cryptomilk, 2019-01-21:
If the 3600 is a new official firmware and the rrlogd_patcher can correctly patch it, you should open a bug at the rrlogd_patcher project. Else you wont get it fixed ...

TinusScheppers, 2019-01-26:
@Maja61
> 
> 
> > @Maja61 Yes, as the README.md states clearly: Gen1 Firmwares >= 3514
> 
> Update the Robo to 3600 and again follow the guide of the patcher, but again still error 500.
> 
> EDIT: seems to be a problem with chrome, with Firefox no error 500 and map still available during charging.

I tried the same but I still get the error (in chrome, in firefox, ...).
I also opened a new issue [here](https://github.com/JohnRev/rrlogd-patcher/issues/3).

How did you get it working?

TinusScheppers, 2019-01-26:
> @Maja61
>
>
> I tried the same but I still get the error (in chrome, in firefox, ...).
> I also opened a new issue [here](https://github.com/JohnRev/rrlogd-patcher/issues/3).
>
> How did you get it working?

Sorry, I ran a new (short) cleaning session and after docking the error came once and then it started working!

hyperbit00, 2019-02-09:
Patcher Works with Xiaomi 1. Gen  and Firmware 3600

No 500 failure at the MapsTab.
But make sure you run the robo one time ;) otherwise there is no fresh readable map.

https://github.com/JohnRev/rrlogd-patcher

josch, 2019-02-11:
I can confirm that the rrlogd-patcher makes the error 500 during charging disappear.

To close this issue, maybe the initial error 500 before the first cleaning should be replaced by a message saying that to see the map, one needs to have vacuumed at least once.

dymn, 2019-02-14:
No, Xiaomi 1. Gen and Firmware 3600 - still 500 error after the first cleaning too.

josch, 2019-02-14:
With gen 2 I also am getting error 500 in cases, where the robot map gets messed up due to the robot getting carried around. I was not able to reproduce this issue reliably yet.

cryptomilk, 2019-02-14:
@dymn Did you build the firmware based on this howto: https://github.com/dgiese/dustcloud/wiki/Cloud-Free-Firmware-Image-With-Valetudo

dymn, 2019-02-14:
@cryptomilk No, I installed version 3600 with root, and via rrcc installed Valetudo. Then patched rrlogd using rrlogd-patcher, and installed it in the system, restarted. The patched rrlogd has md5 - 18ac152fcb105d81f7270ce5d8afc226

Billy3195257, 2019-03-04:
I patched rrlogd on 3600 Gen1 successfully. Error 500 disappeared.
Don't forget to make a new file executable:  chmod +x rrlogd
Switch off/on, press "go to dockstation". Map is shown during charging as well!

closed 2019-04-06 by Hypfer