Hypfer/Valetudo #41: Sometimes only a part of the map is generated

danielpondruff, 2018-11-05:
Sometimes it happens that not the whole map is generated, but only a part of it.

I have attached a video to it. 
https://youtu.be/yGNcIR5bgXI

The problem occurs not only in Homeassistant (set up as a camera), but also when I access the maps directly via URL. After a few refreshes only half of the image comes up every now and then.

danielpondruff, 2018-11-21:
any update here ?

kgc210, 2018-11-21:
I have the same issue but I don't think this is a bug in Valetudo it's self, more how Home Assistant renders the image.
The reason I say this is that if you download every png that gets generated in the tmp folder they are all complete images.

leonardpitzu, 2018-12-03:
i have the same issue - it's Valetudo as i use it to control the vacuum directly (no home assistant integration, etc.)

danielpondruff, 2018-12-03:
> I have the same issue but I don't think this is a bug in Valetudo it's self, more how Home Assistant renders the image.
> The reason I say this is that if you download every png that gets generated in the tmp folder they are all complete images.

Nope. this is not Home assistant releated. I have this issue also when i get the map with my webbrowser directly from bei vacuum. 

> i have the same issue - it's Valetudo as i use it to control the vacuum directly (no home assistant integration, etc.)

i thinks this is a valetudo issue, too

leonardpitzu, 2018-12-12:
there seems to be some size limiting thing - during cleanup the map is correctly generated. after a certain area is covered a new map is generated. i guess it has to do with the new map format discussed in #44 - the map has to be pieced together so we can see the entire map...

ideasman69, 2019-01-19:
I've noticed the same as @JohnRev. Like in this picture, the areas highlighted in red have been completed and did have data earlier, but now has disappeared...

[image]

I just rooted my gen1 yesterday accoring to [this guide](https://hackmd.io/s/HkwF33oWE). Firmware v3514, no errors from firmwarebuilder and patching of rrlog says it was successful.

vevsvevs, 2019-03-07:
I've got the same ripped image only with `returnImage` parameter, but never with a direct link to png.

closed 2019-04-06 by Hypfer