Hypfer/Valetudo #42: Timezones

stefanct, 2018-11-12:
The roborock firmware uses its own timezone configuration: even if you configured the timezone of the ubuntu system correctly and `date` shows the right time the firmware still uses its own tz.
The timers of the roborock (stored in `/mnt/data/rockrobo/apptimer.cfg`) that are also set by valetudo's timer configuration dialog abide this tz and will happily wake you in the middle of the night (been there, done that).

To fix this you have to set two files in `/mnt/data/rockrobo` correctly.
For example, to set the timezone to `Europe/Vienna` you can execute this via ssh:

    echo 'Europe/Vienna' > /mnt/data/rockrobo/timezone
    ln -sf /usr/share/zoneinfo/Europe/Vienna /mnt/data/rockrobo/localtime

Alternatively, `mirobo` does also the correct thing (although wasting a few bytes):
    mirobo  --ip ... --token ...  timezone Europe/Vienna

It would be very nice to be able to do that within valetudo or at least display the timezone information on the timer page ;)

LemarinelNet, 2019-01-07:
Hi
I'm also a victim of the sudden-in-need-of-cleaning-in-middle-of-the-night robot !
Set the timer last week, and this morning, the robot woke up at 6h30, instead of 13h30...
I've now setup timezone info, but the "date" command still return the wrong time ? Is a reboot needed ?

stefanct, 2019-01-07:
The system settings are distinct from those used by the xiaomi software. To set the timezone of the system you can use `dpkg-reconfigure tzdata` as on ordinary Ubuntu machines. I presume you did not change that yet? Cf. `cat /etc/timezone`.

stefanct, 2019-01-07:
Well, that's from dustcloud not Valetudo... so yes, if you know about the issue before changing the firmware that's the easier way to do it but being able to change it later makes sense nevertheless IMHO.

closed 2019-02-16 by Hypfer