Hypfer/Valetudo #43: MQTT Support

Hypfer, 2018-11-13:
* Add the possibility to configure a mqtt server 
* Provide the current status via mqtt
* Enable commands via mqtt
* Provide the latest map via mqtt

Hypfer, 2018-11-13:
There are some standards which enable mqtt device autodiscovery.

Evaluate those and implement one? All?

Landrash, 2018-12-16:
@Hypfer Most likely a MQTT server on the device is overkill.
The 3 others seems like a great idea and opens up quite a few possibilities. :)

Auto [discovery](https://www.home-assistant.io/docs/mqtt/discovery/) implemented by Home Assistant would be a good option but I'm not aware of any others. Currently the MQTT vacuum doesn't have this support but it could be implemented as a switch, sensor and camera.

Hypfer, 2018-12-16:
Nah I wasn't planning on running a server on the vacuum. Just a way to configure the Server IP/Hostname it should connect to 🙃

diabl0w, 2019-01-01:
this would be a great easy way to implement the vacuum into almost any existing smarthome environment

closed 2019-03-22 by Hypfer