Hypfer/Valetudo #44: Support for the new persistent map format

Hypfer, 2018-11-20:


leonardpitzu, 2018-12-03:
nvmap attached in new rr.gz format

[navmapfirst.ppm.log.rr.gz](https://github.com/Hypfer/Valetudo/files/2641039/navmapfirst.ppm.log.rr.gz)

JohnRev, 2018-12-10:
Some findings:

The static encryption key (`RoCKR0B0@BEIJING`) is no longer used.
Instead, a random key is generated for each encrypted file. The key is sourced from `/dev/urandom`.
Then, this key is itself encrypted using `mbedtls_pk_encrypt`. The first 512 bytes of the file represent the encrypted key.
The remaining file content past the first 512 bytes is the map encrypted with `mbedtls_aes_crypt_ecb` and the random key.

That's what I have found so far, but it would be cool if someone else can independently confirm this.

JohnRev, 2018-12-11:
Here's another update.

The encrypted `rr.gz` files go through this process:
1. A random key is obtained by reading `/dev/urandom/`
2. The key is encrypted with an RSA public key, using `mbedtls_pk_encrypt`
3. The encrypted key is stored in the file header (first 512 bytes)
4. The gzip-ed file is encrypted using AES ECB  `mbedtls_aes_crypt_ecb` and appended to the `.rr.gz` file

This is presumably so that Xiaomi can later decrypt the key using their RSA private key, which will then enable them to decrypt the gzip archive.

Some possible workarounds:
1. Intercept the calls to `/dev/urandom` and redirect to a custom file. This way, we will always know what the "random" key is.
2. Replace the public key with one which we know the private key for
3. 

Disable encryption all together.

Options #2, and #3 would break the compatibility with the Xiaomi cloud servers, and so they are best used with dummycloud enabled. Option #1 could theoretically still work, but can be easily detected by Xiaomi.

I tried experimenting with Option #1 for a bit, but was unsuccessful. So I decided to just disable encryption, since I do not care about the Xiaomi cloud because my vacuum is not allowed to connect to the internet anyway. Turns out this is the simplest option.

I updated to 001780 and patched `rrlogd`. The patched file is attached here.
Then, I updated the regex that I pointed out in an earlier comment:  https://github.com/Hypfer/Valetudo/issues/40#issuecomment-443730275
Valetudo now works perfectly with v001780.

To use the new rrlogd, just download the file from here, copy it to the vacuum, and move it to `/opt/rockrobo/rrlog/`.

Here are the hex patches used:
```
﻿0001051C: 4B 47
00010746: FF E3
00010747: F7 20
```
[rrlogd_patched_001780.zip](https://github.com/Hypfer/Valetudo/files/2667438/rrlogd_patched_001780.zip)

leonardpitzu, 2018-12-11:
@JohnRev can you patch a rrlog (of course i will privide it) for gen2? i would try this and see how the new persistant pams work out...by the way - what's the setting needed to enable the map storage feature?

JohnRev, 2018-12-11:
This is gen2. I should have clarified

leonardpitzu, 2018-12-11:
gr8. i'll download 1780 and get myself to work. i guess we should e carefull and stop the automatic update mechanism - xiaomi will most probably close some doors and we might remain stuck with some default stuff...

JohnRev, 2018-12-11:
My vacuum has never connected to the internet, so I'm not too worried about the automatic updates.
If you do use the patched rrlogd file, it would probably be wise to use dummycloud too as I'm pretty sure the patched rrlogd breaks compatibility with the xiaomi cloud services.

leonardpitzu, 2018-12-11:
i was connected to the internet (i got the gadget exactly when it came on the market). now i am on dummycloud and ccrypt is removed (just an empty file). my firewall is also plugging all holes so i should be safe. have you tried to enable the map storage feature? and if yes how did you do it?

JohnRev, 2018-12-11:
No, sorry, I haven't looked into the map storage feature. Tho if you do figure out how to enable it, let me know!

leonardpitzu, 2018-12-11:
@JohnRev i updated the beast and used your patched rrlogd but i keep getting error 500 from the webserver...

JohnRev, 2018-12-11:
@leonardpitzu, you will have to compile Valetudo with the updated regex from https://github.com/Hypfer/Valetudo/issues/40#issuecomment-443730275

leonardpitzu, 2018-12-11:
Already done that - I get the correct path to the files and I am trying correctly to open them. That’s where I got with the debugging.

JohnRev, 2018-12-11:
Interesting. It works fine for me. How does your `/mnt/data/rockrobo/rrlog/` folder look like?
If there are a bunch of subfolders there, i suggest you delete them and try again with a fresh sweep of the vacuum.
Then, after the vacuum is docked, there should be a new subfolder containing a bunch of .gz file. Can you confirm they are unencrypted gzip files?

JohnRev, 2018-12-11:
Oh, and did you reboot your vacuum to make sure it uses the patched rrlogd?
If you dont want to reboot, try killing the rrlogd process. The vacuum will respawn one with the patched rrlogd (assuming it is placed in the right path)

dugite-code, 2018-12-12:
@JohnRev How did you go about patching rrlogd?

leonardpitzu, 2018-12-12:
@JohnRev i've done some more debugging (i know linux quite well but i'm 250km away from the vacuum so debugging is a bit more tricky :) ). anyway here's what i found (again, usinf fw 1780, your patched rrlogd and a patched/modified version of valetudo):

- the webserver searches for the long folder names stored in /mnt/data/rockrobo/rrlog
- the webserver makes a list of folders and searches for maps (this time using the new regex from #40)
- if a folder and a mapfile and a logfile are found within the folder the map is being "reconstructed", that is the log is being decrypted.

and here it stops with the error: > digital envelope routines:EVP_DecryptFinal_ex:bad decrypt <

previousely the log was being encrypted with the standard symetric key which we all know. so the webserver was decryptiong it using the key, unpacking it and do the rest of it's thing.
From your description o figured you disabled the encryption all together so the webserver now failes to decrypt the data. Even more, when i download the encrypted logs locally and try to unpack themn it fails - probably the magic number is wrong.

JohnRev, 2018-12-12:
@leonardpitzu :Just to confirm, did you make sure the running rrlogd process is the patched one? Simply replacing the file won't do it.

leonardpitzu, 2018-12-12:
@JohnRev copied the patched rrlogd and rebooted the thing... i'm doing some more testing to see what's going on...

JohnRev, 2018-12-12:
Oh! I think I attached the wrong rrlogd. In the one you're using right now, the first 512 bytes are the encrypted key and the rest of the file is the gzip.
I have updated my initial comment. If you don't mind, please use this rrlogd. It is the same one as I have on my vacuum right now.
md5sum is `93e8df884b4c6162ffa2a11161f47429`
Send it to the vacuum, kill the running rrlogd process (or reboot), and try again.
[rrlogd_patched_001780_gen2.zip](https://github.com/Hypfer/Valetudo/files/2670885/rrlogd_patched_001780_gen2.zip)

JohnRev, 2018-12-12:
@dugite-code: I disassembled `rrlogd`, looked for the function call that appends the encrypted key to the header, and disabled (`NOP`) that call. Then I looked for the call that stores the encrypted gz bytes to the file, and made them store the unencrypted buffer instead. The hex patches posted in my first comment do just that, and you can apply them to your unpatched rrlogd (of version 1780) to get the same patched rrlogd that I posted here.

leonardpitzu, 2018-12-12:
@JohnRev i looked at the attached rrlogd and it's not the patched one. I patched it and am testing it now. will report back

JohnRev, 2018-12-12:
Thanks! I am waiting. Sorry for attaching the wrong `rrlogd` initially ... I got my files mixed up :smile:

leonardpitzu, 2018-12-12:
@JohnRev sudo "you diserve a beer!" it's working!

JohnRev, 2018-12-12:
Awesome!! And thanks for being willing to testing it :smile: 

I discovered there's some 500 errors still showing up after some time, but these are strictly related to the code logic of how the `/mnt/data/rockrobo/rrlog` is scanned in `FIND_LATEST_MAP_IN_ARCHIVE` function. I will write about them in #40 .

Glad it's working, and hope this helps others too!

JohnRev, 2018-12-14:
@leonardpitzu, I have a question for you.
After rebooting the vacuum, does the robot forget the last map when performing a new sweep and generates a new one?

leonardpitzu, 2018-12-14:
@JohnRev mine also has Alzheimer and generates a new map. This is a "feature" i would guess unless we figure out what the "save map" button actually does. If we can set the same flag, by hand, we can keep the map between reboots. I read somewhere that the setting is some new file in /mnt/data/rockrobo. The file has to contain the value 1 in order to enable the map storage feature. Yet i can't seem to find that post nowhere so i am stuck right now...

JohnRev, 2018-12-14:
I think I was able to get the saved map functionality to work. Would you be willing to test it?
1. Do a full sweet to generate the map to be saved
2. Install python-miio `pip install python-miio`, then execute the following (replacing with the correct variables, obviously).
3. Execute:
{code}
export MIROBO_IP=vacuum_ip
export MIROBO_TOKEN=vacuum_token_key

mirobo raw-command "set_lab_status" 1
mirobo raw-command "save_map
{code}
4. Reboot the vacuum
5. Do a new sweep

JohnRev, 2018-12-14:
Indeed, the above commands seem to produce a file `/mnt/data/rockrobo/lab.cfg` with content 1.
Not sure if just creating that file (`echo 1 > /mnt/data/rockrobo/lab.cfg`) would suffice, but the above commands worked well for me, as far as I can tell.

leonardpitzu, 2018-12-14:
@JohnRev that was the config file + flag. I am not sure though how this should work... Between reboots the maps is lost but i still get to see the track/path of the vacuum (well, at least the last part of it as it's split as o lately on more pieces). I have a persistent gridmap in /dev/shm (around 1MB) but even witht the "preferGridMap":true i only get to see the path...

JohnRev, 2018-12-14:
Note sure I understand. Do you mean in Valetudo you see the path but without a map?

leonardpitzu, 2018-12-14:
yes

JohnRev, 2018-12-14:
Now that is something I haven't seen yet. It would be interesting to see which path file is being parsed by Valetudo.
Using which method did you enable the persist flag? Manually in `lab.cfg`, or with the mirobo `set_lab_status` and `save_map` commands?

leonardpitzu, 2018-12-14:
i used lab.cfg manual edit. valetudo is selecting the latest folder and searching for navmap and slam.

JohnRev, 2018-12-14:
Perhaps the `mirobo raw-command save_map` command was needed?
Just tried a reboot and my old map was still there, with the old path.
I did a new sweep, and the robot "remembered" the last map and just started tracing over it

JohnRev, 2018-12-14:
Nice setup!

So now we have a way to use Valetudo on gen2 v001780 (with patched rrlogd) and a way to keep maps after reboot (with `lab.cfg`)!!

Hypfer, 2018-12-14:
So they are still using ppm maps?

How are virtual walls stored?

JohnRev, 2018-12-14:
Yes, ppm is still used.

I haven't been able to figure out where virtual walls are stored (yet), and I don't have the mi app to test it either (my vacuum isn't connected to the internet).

It would be helpful if someone who does use the mi app can provide some insight

dugite-code, 2018-12-15:
Hey could you post this to the  https://github.com/marcelrv/XiaomiRobotVacuumProtocol github? I know the miio guys check there for newly discovered commands.

Hopefully someone will figure out the virtual walls soon

JohnRev, 2018-12-15:
Posted it there https://github.com/marcelrv/XiaomiRobotVacuumProtocol/issues/15#issue-391360556 with some additional info

JohnRev, 2018-12-15:
@Hypfer : These are *possibly* stored in `/mnt/data/rockrobo/user_map0` and/or `/mnt/data/rockrobo/PersistData*.data`.

leonardpitzu, 2018-12-15:
@JohnRev after the scheduled reboot (every day my vacuum reboots on its own at 3:30am) the maps were not visible any more in valetudo - again only the last part of the track was visible. i removed dummycloud and checked with the official app - the map is saved correctly and all settings are as well correct. I made a full factory reset of the device and made a sweep. Maps were saved (persistent files were generated). After a reboot, boom, maps were gone again in valetudo. I am out of ideas...

JohnRev, 2018-12-15:
Hmm... I haven't faced this issue yet, and I have been rebooting my vacuum daily.
Are you getting error 500 in Valetudo? Maybe the issue is from what I described here: https://github.com/Hypfer/Valetudo/issues/40#issuecomment-446504197 ?

herrwusel, 2018-12-15:
@leonardpitzu & @JohnRev Why do you reboot your device daily if I may ask?

JohnRev, 2018-12-15:
For some reason that I haven't been able to figure out yet, my vacuum loses connection to the internet overnight (but otherwise remains functional), and so  I must reboot it to get it connected to my router again

leonardpitzu, 2018-12-15:
@JohnRev done that. No change at all. What those commands do is identical to what the original app does. I've done the same settings with both and still the same behaviour. I also encounter #40 but i overcome it by removing the new folder. Still, the map is not visible from the "old" folder - only the track (see attached screenshots - one is valetudo, the other is the original app. both screencaps are after a manual reboot [sudo shutdown -r now]).
@herrwusel i'm not rebooting it - it does so on its own at 3:30 every morning. It's some sort of scheduled tasks. I would assume all vacuums do so but you don't see it because either you're not monitoring the network or you don't have startup/shutdown/reboot notifications as i do.
[image]
[image]

leonardpitzu, 2018-12-15:
@JohnRev here you go (there are also some new generated folder so you can have an idea what happens @reboot).
[Archive.zip](https://github.com/Hypfer/Valetudo/files/2682862/Archive.zip)

JohnRev, 2018-12-15:
`gzip: navmapfirst.ppm.log.rr.gz: decompression OK, trailing garbage ignored`
After reboot, your navmapfirst.ppm is empty (or corrupt?), and that is why the map is blank and Valetudo can't display it.
The official app probably pulls the map from the Mi cloud, and that's why it's displayed (just a guess)

I don't know why your navmapfirst.ppm.log.rr.gz is like that ...

leonardpitzu, 2018-12-15:
@JohnRev user_map0 is 198k. i'll try a fsck... the map is resaved. i made a new one this morning, rebooted and valetudo is what you see it above...

leonardpitzu, 2018-12-15:
@JohnRev another idea - if i delete all folders would it bring something? what else do i need to refresh to make everything "like new"? factory reset brings nothing - i've done it and everything is still there

JohnRev, 2018-12-15:
I have once deleted all the folders inside the rrlog folder to refresh things after a factory reset. You could try that, but I'm not sure it's related.

leonardpitzu, 2018-12-15:
@JohnRev at what level have you set the logging? Is upload still enabled? are the other scripts still running or do you have them exit as soon as they are called?

JohnRev, 2018-12-15:
I don't recall messing with the logging levels and the scripts (they don't exit). I do have tje xiaomi servers in the hosts file and the iptables rules in rc.local (from dummycloud setup)

leonardpitzu, 2018-12-16:
@JohnRev a full wipe and many hours later i got the damn thing to have consistent logs - the rr.gz are extractable and have no errors. still they are empty and the map is as well empty - i only see the path of the vacuum as in the screenshot above. again i am out of ideas...

Hypfer, 2018-12-17:
I think that there might be no "new map format" 🤔
Maybe the virtual walls are stored the same way as zones are stored.

Hypfer, 2018-12-17:
https://github.com/marcelrv/XiaomiRobotVacuumProtocol/issues/15#issuecomment-447636171

I guess you can fetch them with a miio command? Or is that write only?

cryptomilk, 2018-12-17:
@JohnRev thanks for your work. I've added support for patching rrlogd to the imagebuilder script of the dustcloud:

https://github.com/dgiese/dustcloud/pull/162

leonardpitzu, 2018-12-18:
@JohnRev my partition was bad. i had to kill almost all services to be able to unmount the partition. i did so and in the end i run fsck. now i have (again) a normal behaviour where maps are stored in between reboots. the patched rrlog + fw 1780 work great.

cryptomilk, 2018-12-22:
FYI: The latest 0.9 valetudo release works with v001780 and a patched rrlogd quite fine.

I wonder how the persistent map feature really works. Normally I guess you want to store a complete map of your rooms as persistent and be able to load it. The robot might do a run and might not able to enter a room because the door is closed, however you still want a map with all rooms mapped so you can send the robot for a zoned clean up.

You can activate the persistent map feature doing the following:

`echo -n 1 > /mnt/data/rockrobo/lab.cfg`

Time for more investigation.

cryptomilk, 2018-12-23:
I've patched v001792.

```
radiff2 rrlogd.v001792 rrlogd.v001792.patched 
0x000105a8 4b => 47 0x000105a8
0x000107d2 fff7 => e320 0x000107d2
```

```
radiff2 -a arm -D rrlogd.v001792 rrlogd.v001792.patched 
(locked: no new keys can be created (anal.split))
--- 0x000105a8  4b
- add r0, sp, 0x12c
- movs r2, 0x10
+++ 0x000105a8  47
+ add r0, sp, 0x11c
+ movs r2, 0x10

--- 0x000107d2  fff7
- bl 4294966930
+++ 0x000107d2  e320
+ movs r0, 0xe3
+ invalid

```

Here is support for patching rrlogd directly when building the image (uses bsdiff/bspatch)

https://github.com/cryptomilk/dustcloud/tree/master-imgbuild-rrlogd

JohnRev, 2018-12-30:
>  Maybe the virtual walls are stored the same way as zones are stored.

After spending a ridiculous amount of time reverse engineering the saved map formats, I think I have finally figured it out!!
The no-go zones and virtual wall coordinates are stored in : `/mnt/data/rockrobo/PersistData_1.data`.

The file first needs to be zlib decompressed, and inside it it contains a block of fixed size where virtual wall coords are stored, and also no-go zones.
The fixed size of these blocks is why the xiaomi app has an upper limit of virtual walls that you can create (50 or so, I think).

I need to spend a bit more time writing some codes to parse these files. This will hopefully allow us (with some effort) to have the virtual walls and no-go zones managed and displayed from within Valetudo!

[33 more replies]

closed 2019-01-03 by Hypfer