Hypfer/Valetudo #58: Unreliable Coordinate System

ftastisch, 2018-12-17:
Hi!

First of all: Thanks for providing such a good tool for the Xiaomi vacuum. I like it very much!

One thing I noticed after playing with Valetudo for a few days: The coordinate system seems to be kind of unreliable.
Standing in front of the charging station (looking at it), I had a reproducible movements with -x towards me and +y to my left.
After setting up my zones accordingly (did it in the config.json directly), I had to reboot my vacuum to see them and the coordinate system is completely different, an no matter how many reboots and whatelse, I can't get it back the way it was. Now it is +x towards me, and -y to my left (so rotated by 180°, if I'm not mistaken). I did not move the charging station in the whole process.

Additionally, neither orientation fits the description under "Zoned Cleanup" with x for left and right. ;)

Please let know if I can help in any way to solve this.

Best regards,
Fo

ftastisch, 2018-12-17:
Ok, it is very weird. I changed all the signs for my zones, in order to correct them and tested it with one zone (worked perfect, btw, so again thanks for your awesome work ;)). After another reboot though, the coordinate system is back in the previous state. It seems to be completely random...

Hypfer, 2018-12-17:
It's most likely not a bug in valetudo.
I guess this might be fixed when persistent map support is available

maxpautsch, 2018-12-18:
This is due to the concept how the map is generated by the vacuum.
You can fix your problems by locating the dock on a straight wall with enough free space on the left and right side.

ftastisch, 2019-01-13:
As this is not an issue in valetudo and updating to 3514 improved the situation, I would close this issue.

closed 2019-01-13 by ftastisch