Hypfer/Valetudo #68: Vaccum drives back home after Zone Cleaning

hagenuck1, 2018-12-26:
Hi, the roborock is driving back home, when using the new "roborock/zone" url. I'm trying to find out all coordinates and cleaning some rooms for this, so the map is completely recognized. 
But as soon, as the robot has finished cleaning a zone it goes back to the charger.

This isn't interruptible, even if I select a second zone or a new "goto" location.

Hypfer, 2018-12-26:
I don't think that valetudo can change anything about that :/

hagenuck1, 2018-12-26:
Okay. So it‘s not possible for Valetudo to cancel a task, too. Right?

PBue, 2018-12-26:
Hi,

something similiar was reported as part of #35.
I fixed part of the problem, but the vacuum still only cleans the first zone of an area and then returns home.

I am using a Gen1-vacuum.

hagenuck1, 2018-12-26:
Thanks for the reply. 

If I see it right, sending the stop parameter to the vacuum should stop it instantly and you should be able to send a new command. This is being used in the scripts of some users in the home assistant community. 
Will give that a try tomorrow.

hagenuck1, 2018-12-27:
So, just tried the following:
1. try:
- started a zone cleaning using the /zone url. 
- Pressed „stop“ in Valetudo while ist’s cleaning the first zone
- Stopps cleaning
- Assigned a new zone
- Roboter drove to the new zone and starts cleaning

2. try:
- started a zone cleaning using the /zone url
- Assigned a new zone while Roboter clans 
- Roboter continues to clean the first zone

3. try:
- started a zone cleaning using the /zone url. 
- waited till the Roboter finished cleaning the first zone and automatically starts driving back to home
- (stop button is now grayed out)
- Pause driving back by pressing pause button in Valetudo
- assign a second zone 
- Roboter drives to the second zone and cleans it. 

4. Same as 3. try, but without the pausing step
- Roboter still goes to the second zone and cleans it. 

My suggestion would be:
- if you assign a new goto or zone using the /zone url while Roboter is driving home or cleans a zone there should be a menu to cancel current task and clean the new assigned zone. If you press to cancel it should send a pause command and then the new zone. 

Greets

PBue, 2019-01-13:
> > I fixed part of the problem, but the vacuum still only cleans the first zone of an area and then returns home.
> 
> Can you see the rectangles added to an area in the config.json file? I just tried it and adding zones does not work for me (valetudo 0.0.9, firmware 3514).

Release 0.0.9 does not yet include the patch :-)

closed 2019-04-28 by Hypfer