Hypfer/Valetudo #69: Easy way to store and restore Map for Gen1 & Gen2

ovaltineo, 2018-12-29:
I'm not sure if anyone has done this before (maybe I didn't look hard enough) but it was pretty easy to store and restore the current map on a rooted Xiaomi robot.  To store, I just copied /mnt/data/rockrobo/* (excluding directories) to a backup folder.  To restore, I copied from the backup folder to /mnt/data/rockrobo, then I kill 'player' process which gets restarted by the watchdog within a minute.  Voila, the robot now starts using the restored map.

I think these two functions can easily be incorporated in Valetudo.  It certainly saves me from re-mapping the house whenever the robot decides to forget the map.  My zoned cleanup coordinates are always spot on because of this.

roflcoopter, 2019-04-13:
> mnt/data/rockrobo

Are you doing this consistently to have separate maps? 
Are you deleting the contents of /mnt/data/rockrobo/ and then swapping in different maps?

ovaltineo, 2019-04-13:
I am using it to restore the map whenever the robot gets stuck and someone returns it manually to the base.  If there are two saved folders, then map swapping should work.  I don't need to delete the files because they are copied over from the backup.  I use a remote SSH script to do the copying.  This script is triggered via Google Home voice command configured in IFTTT.

roflcoopter, 2019-04-13:
Interesting! Mind sharing your script?

ovaltineo, 2019-04-14:
Here's the script run on my raspberry pi for copying the files from backup.  ssh key is used so no there are no password prompts.  I use Blynk to trigger this script.  IFTTT sends message to Blynk.

##restore_rockrobo.sh
MIROBO_IP=192.168.1.15
ssh root@$MIROBO_IP cp /mnt/data/rockrobo/backup/* /mnt/data/rockrobo;
ssh root@$MIROBO_IP sync
ssh root@$MIROBO_IP reboot

TheMicroDevil, 2019-04-24:
Just the hint i was looking for, thanks!
I've  developed a tool in bash, which can issue single/multiple zone cleaning (running on device or wherever). This could be the missing function to reset the map before start.

To be mad,  theoretically it could then support multiple floors too, if needed.
Something like:
clean $zone@$floor
- if $floor not $standard_floor
- play sound (hey, you have to do something)
- wait for error 3
- you take the robot to $floor, error 3 => 'Wheel floating', sound stops
- wait for error 0 - no error, the robot is there
- cp /mnt/data/rockrobo/backup/$floor/* /mnt/data/rockrobo
- restart player
- start cleaning $zone@$floor
- $standard_floor is now $floor

ovaltineo, 2019-04-25:
Yes, it can be used to remember multiple floors.  The only problem is that the map seems to have an expiry date -  I have to do a full clean and replace my backup when this happens.   The expiry is over a month ... maybe 6 weeks.

hagenuck1, 2019-06-13:
I'm using this for a week now, too and have the following additions:

last_map, ChargerPos.data, PersistentData_1.data (& PersistentData_2.data; Only exists if there are NoGo.Zones configured) are the only needed files.

If I copy my map to the robot the following files MUST be deleted if they exists:
PersistData_2.data, StartPos.data, user_map0

Then the files have to be copied and overwritten to that folder. 
Now the robot has to be rebooted using "reboot" with SSH.
(Restarting valetudo and/or rrwatchdoge service isn't working great for me.)

I configure the NoGo-Zones using RRCC.

If the robot destroyed the map in the current/ last clean it changes RoboController.cfg:
need_recover_map = 0 to 1
- This has to be fixed BEFORE Restarting, otherwise the copied map will be deleted and it creates a new one.

patonadig, 2019-07-15:
Hi, I'm very interested in the multiple floors feature. What if instead of having multiple maps we would have a single one with all the floors? The floors could be mapped individually into different files and later manually merged into one (if feasible). If required the floors could be represented as adjacent or with a fake corridor / door in between, and of course the robot would fail to go from one to the other, but when placed in any of them it should recognise the map. It would also fail to dock. However I don't know if this map manual edit / merging is possible. Any suggestions?

ovaltineo, 2019-07-16:
Why would you ever want such a complicated solution (pretty sure it won't work) when the one backup directory per floor is proven to work?

patonadig, 2019-07-16:
The advantage could be that once the map is saved there is nothing else required to do. No map switching needed anymore. To be honest I haven't still tested the backup folder method but i'll do soon.

Hypfer, 2020-05-15:
stale.

Needs to be discussed in a new issue

closed 2020-05-15 by Hypfer

fokcuk, 2021-05-28:
hm... its 2021 and the map saving functionality is still missing. Any updates?

joseska, 2021-06-29:
I´m waiting for this functionality too......
for now I´m using this "https://github.com/Thyraz/MapLoader" and Node-Red with a Gen1 vacuum.....