Hypfer/Valetudo #71: Timers not working on Gen2 robots?

jansohn, 2018-12-31:
I added the following cron expression (`0 0 10 ? * MON,THU *`) and expected my robot to start cleaning this morning but nothing happened. I'm using image v11_001518 and Valetudo 0.0.9.

Is this working for someone?

Hypfer, 2018-12-31:
That's unsupported Cron syntax.

You can find some working examples here:
https://python-miio.readthedocs.io/en/latest/vacuum.html

closed 2018-12-31 by Hypfer

axel-kah, 2018-12-31:
It's also a good idea to check the current date set on your robot. Mine was off by at least half a day. Depending on your setup you can configure a ntp server running in your LAN or on the internet.

jansohn, 2018-12-31:
OK, I had it generated by http://www.cronmaker.com/ which seems to produce a different cron format. Would still be nice to have some sort of guidance and feedback when entering the timer value.