Hypfer/Valetudo #73: Calling Zone via HTTP Request

kaypohl, 2019-01-02:
Hi,
is there any possibility to call a zone cleaning via http request / api?

Kay

matthiasharrer, 2019-01-02:
You need to send sth. like `[[-1000, -1000, 1000, 1000, 1]]` via HTTP PUT to `/api/start_cleaning_zone` for cleaning 2x2m around the charging station one time.

See also:
https://github.com/Hypfer/Valetudo/blob/master/webserver/WebServer.js#L243
https://github.com/Hypfer/Valetudo/blob/master/miio/Vacuum.js#L384

kaypohl, 2019-01-02:
Hi and thanks for your reply. I will try that. I thought it would be possible to "trigger" the already in the webfront configured zones?

Kay

matthiasharrer, 2019-01-02:
I think thats currently not possible directly, but you can retrieve the configured zones at `/api/get_config` and then figure out the coordinates of the corresponding zone.

matthiasharrer, 2019-01-03:
I think "ok" is the correct response .. is the robot reacting to the command?

kaypohl, 2019-01-03:
No, he did not. But it was my fault. I had to change the Body format to "json" and not to "text". With json it works. Thanks for your help.

closed 2019-01-03 by kaypohl