Hypfer/Valetudo #75: API Documentation

diabl0w, 2019-01-06:
Is there a place that documents which API's are available and how to use them? I see a lot of people using http requests for various API's and I don't see how they know how to use them. Are we just supposed to look at webserver.js? 

Unfortunately, these things are more of a hobby for me and so I don't know enough to understand the coding in webserver file to figure out the API's

Hypfer, 2019-01-06:
The HTTP APIs are not intended to be used by anything else than the Webinterface itself hence there will be no documentation.
At some point there will be #43 which solves this.

closed 2019-01-06 by Hypfer

Hypfer, 2019-03-22:
MQTT is now available

https://github.com/Hypfer/Valetudo/releases/tag/0.2.1