Hypfer/Valetudo #80: SSID gets lost every morning

willchr, 2019-01-17:
Hi together,

after installing and preparing the robot (1780, Gen2, valetudo0.0.9) 4 days ago actually everything went nice. Maps are persistent and miio is working moderatly, means, it's reporting a lack of communication to the dummycloud from time to time, however after 5 sek. the system is responsible via valetudo again.

My problem is, that every morning the robot isnt reachable anymore. I have to set back it to 192.168.8.1 and redefine SSID/PW in valetudo again.

Does anyone had similar experiences? What can I do?

Greetings
Christian

dugite-code, 2019-01-17:
@willchr I've been having similar issues only it wasn't every morning for me, it was at random times of the day. I haven't seen the issue since I disabled logging using https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum/scripts/disable_logging.sh

That said I only did this 3 days ago so take that with a grain of salt

willchr, 2019-01-17:
Thanks! Have disabled the log update and will give a feedback in a couple of days.

willchr, 2019-01-19:
unfortunately unabling of logging according this link (using https://github.com/dgiese/dustcloud/blob/master/devices/xiaomi.vacuum/scripts/disable_logging.sh) didnt help. And yes, the wifi-connection gets lost not every morning but at random times :(.

@dugite-code: how did you define the wifi-params? via valetudo-webfrontend?

dugite-code, 2019-01-21:
I just defined it using python-miio, simply because I had it already set-up on my phone

```
mirobo configure-wifi **SSID** **PASS** --timezone** COUNTRY/CITY**
```

willchr, 2019-01-21:
Thx, I am going to test the local ntb first. Yesterday evenings try with connecting to a fresh installed dustcloud brought the same issue. This morning no wifi-connection again. So the issue seems not being related to the dummycloud itself.

willchr, 2019-01-23:
The local ntb didn't help neither. Yesterday I recognized, that even in unprovisioned mode (192.168.8.1) the Accesspoint gets off after while. Is this normal?
I have installed a cronjob yesterday evening with the task to write a timestamp into a external file (NAS) every single minute. At least I am able to understand with this way, how often and when the wlan-connection gets broken. This night he loses the connection exactly 4AM.

Hypfer, 2019-01-23:
The Access Point will be disabled after a certain amount of time since the last reboot. Xiaomi added this feature to prevent people from hijacking unprovisioned robots.

[https://twitter.com/dgi_DE/status/1072542781527216129](https://twitter.com/dgi_DE/status/1072542781527216129)

Hypfer, 2019-01-23:
Also, the robot reboots every night (at 4AM iirc). This is intended behavior.

willchr, 2019-01-24:
I'm a bit frustrated.
As a dirty workround for the moment I let the robot check the connection status  every 5 min  via cronjob in order to restart the wifi-connection automatically and immediately when the issue occurs.

ideasman69, 2019-01-24:
> *Edit2: Also have a look at this recent commit regarding dummycloud `This creates a successful connection to dummycloud. With this the wifi led stops blinking.` [dgiese/dustcloud@67d7088](https://github.com/dgiese/dustcloud/commit/67d7088a4602c479e2c61f980e11ef5fbfde4ece)

Thanks so much for linking to this. I had the constant flashing of the wifi LED (which didn't bother me) but made the vacuum really unreliable to communcate with. Using both Home Assistant and valetudo were sluggish until I put this fix in. Thanks to @cryptomilk for making the change and @dugite-code for pointing it out!

willchr, 2019-01-24:
I am wondering about the reasons why the robot reboots from time to time. Maybe it's sufficient to prevent it from rebooting, i.e. with apt-get install molly-guard...

hagenuck1, 2019-01-29:
Using 1792 with the same issue currently. But I guess I found part of the issue. Its caused by the „start_wifi.sh“ which is also run after the reboot in the night. 
I changed the logging behavior yesterday to see if the script is even run in the night and it is, but it can’t find the device „wlan0“

Start: 2019-01-29_02:14:12
Stopping AP...
Killing leftover AP daemons...
hostapd: no process found
Starting WPA supplicant background process...
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Could not read interface wlan0 flags: No such device
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
Requesting IP and network settings from DHCP server...
Cannot find device "wlan0"
End: 2019-01-29_02:17:30

Its connecting if I press the reset button on the device and writing this in the logs:
Start: 1970-01-01_00:00:10
Stopping AP...
Killing leftover AP daemons...
Starting WPA supplicant background process...
Requesting IP and network settings from DHCP server...
End: 2019-01-29_05:31:15

cr0, 2019-02-03:
@hagenuck1 Thank you digging into this as I am experiencing the same issue on 1792. However, I am wondering how your solution works. I am not using the imagebuilder's unprovisioned mode, therefore all the said files should not be on in the final image used on the roborock. This is the case because my `/etc/rc.local` is empty and there is no `start_wifi.sh` script.

Any idea what's wrong here?

cr0, 2019-02-13:
This issue seems to be related to current changes in the xiaomi firmware (see the above twitter link). The robot needs to connect to `ot[t]?.io.mi.com` in order to store a long time token (maybe activate itself). Otherwise the token is changed as soon as the robot reboots and then reconnects to the wifi.
Trying to use dustcloud and redirect traffic to `ot[t]?.io.mi.com` to my local dustcloud.

hagenuck1, 2019-02-20:
As mentioned in the Roborock Facebook group by Roborock about 1720 Image „This new update has fixed an issue that caused wifi to shut down under special conditions“

If 1720 is tested with dummycloud & valetudo this maybe solves this issue.

cryptomilk, 2019-02-21:
I have v001720 running since Tuesday.

hagenuck1, 2019-02-21:
1720 isn't available on the servers anymore or what link you used? The Dustcloud links (example replaced with 1720) aren't working for me.

cryptomilk, 2019-02-21:
I've got it from http://dustbuilder.xvm.mit.edu/

However it looks like the robot is drunk when it is driving around. v001712 is working fine.

dugite-code, 2019-03-26:
Now I am no longer experiencing the memory leak bug https://github.com/Hypfer/Valetudo/issues/39 I am experiencing this at 4am rather than random times.

I'm on a gen1 `v11_003194`

I noticed a similar issue with dummycloud when I had valetudo disabled and I believe I solved it by forcing dummy cloud to start before the networking was brought up using `start on (starting network-interface or starting networking)` in the upstart script. Perhaps valetudo has a similar issue?

I'm changing it to start after the networking has started `start on (started network-interface or started networking)` to see if that's the issue. I'll report back in a few days

dugite-code, 2019-03-29:
Ok it's been a few days and I haven't had any issues with the wifi since I changed the upstart config. I'm not 100% sure why starting just after the file system would cause this issue but It could be OOM killing valetudo and it getting restarted when a lot of things are happening at boot. If anyone else is still experiencing this could they test this upstart config

```
#!upstart
description "Valetudo"

start on net-device-up IFACE=wlan0
stop on runlevel [06]

oom score 1000
exec /usr/local/bin/valetudo
respawn
respawn limit 10 90
```

Hypfer, 2019-03-29:
Does this also work in AP mode? I remember that there are two wlan interfaces but I'm not sure if wlan1 is even used at all.

dugite-code, 2019-03-29:
I didn't realise there was a wlan1 in that case you could just wait for any non-loopback interface: `start on net-device-up IFACE!=lo`

Might cause Valetudo to listen on the wrong interface during provisioning if it's switching between `wlan1` and `wlan0` without a reboot. I'll try to find some time to try and test that.

leonardpitzu, 2019-04-11:
same is happening on 1810. every morning i find the guy dead. i have to power it off and back on and reprovision it...

cryptomilk, 2019-04-11:
1810 is a beta version, what do you expect ...

leonardpitzu, 2019-04-11:
Fair enough. My point was the issue is not fixed. I can downgrade but judging by this thread it would not solve the problem at all.

Hypfer, 2019-04-11:
Did you try the changes to the upstart file proposed by @dugite-code ?

leonardpitzu, 2019-04-11:
Did it this morning. Now I have to wait until tomorrow...

cryptomilk, 2019-04-11:
Some wifi issues have been fixed with v1730. We suggest to use v1748.

v1810 has been released Jan 01
v1730 has been released Feb 01
v1748 has been released Feb 26

leonardpitzu, 2019-04-11:
updated to v1748 + @dugite-code upstart script changes. let's see how this works out

leonardpitzu, 2019-04-14:
v1748 (@cryptomilk thanks!) and @dugite-code upstart script changes did the trick. the led is blinking and i can't seem to figure out why yet the robot works as expected (i can fully control it over valetudo)

Hypfer, 2019-04-18:
I experienced a firmware rollback to the one containing valetudo 0.2.3. Maybe that was caused by the same issue described here?

Hypfer, 2019-04-23:
`start on net-device-up IFACE!=lo` does in fact break valetudo for the unprovisioned robot. However since AP mode also uses wlan0, `start on net-device-up IFACE=wlan0` can be used

fvollmer, 2019-04-24:
It appears like this isn't working for at least one person: https://github.com/Hypfer/Valetudo/issues/185#issuecomment-486445155

imbaimbaimba, 2019-04-25:
Yes I am having similar issues. Newly flashed robot, and Valetudo breaks when adding my WiFi credentials to the robot. 
It does connect to my WiFi and gets an IP that I can access, but it seems valetudo stops getting data, so it just wait for states forever.

Anyone have any ideas?

pmazanec, 2019-04-28:
I'm also having similar issue. Valetudo doesn't start on freshly flashed robot 

```
Apr 28 18:18:35 rockrobo kernel: [  711.967867] init: valetudo main process ended, respawning
Apr 28 18:18:37 rockrobo kernel: [  713.967744] init: valetudo main process (10023) terminated with status 1
Apr 28 18:18:37 rockrobo kernel: [  713.967808] init: valetudo respawning too fast, stopped
```

I can ssh to it and start it manually with `service valetudo restart` 

however, after configuring wifi in valetudo gui, I'm no loger able to ssh and valetudo gets stuck on "loading state". 

I'm using valetudo 0.3.1 on mirobot v1 with recommended v11_003468 fw

Hypfer, 2019-04-28:
@pmazanec how does /etc/init/valetudo.conf look like?

pmazanec, 2019-04-28:
the ssh access was my fault (typo in IP). however, valetudo still doesn't start on boot. the log file is empty. after wifi reconfig and reboot it starts normally.

```
root@rockrobo:~# ps aux | grep vale
root      1978  0.0  0.1   1692   524 pts/1    S+   20:03   0:00 grep --color=auto vale
root@rockrobo:~#
root@rockrobo:~# cat /var/log/upstart/valetudo.log
root@rockrobo:~#
root@rockrobo:~# uptime
 20:03:31 up 1 min,  1 user,  load average: 1.28, 0.50, 0.18
root@rockrobo:~#
root@rockrobo:~# cat /etc/init/valetudo.conf
#!upstart
description "Valetudo"

start on net-device-up IFACE=wlan0
stop on runlevel [06]

oom score 1000
exec /usr/local/bin/valetudo
respawn
respawn limit 10 90
root@rockrobo:~#
root@rockrobo:~# service valetudo restart
stop: Unknown instance:
valetudo start/running, process 2378
root@rockrobo:~#
root@rockrobo:~# cat /var/log/upstart/valetudo.log
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
root@rockrobo:~#
root@rockrobo:~# ps aux | grep vale
root      2378 23.7  5.4  82896 27660 pts/2    Ssl+ 20:03   0:02 /usr/local/bin/valetudo
root      2502  0.0  0.1   1692   524 pts/1    S+   20:04   0:00 grep --color=auto vale
root@rockrobo:~#
```

pmazanec, 2019-04-28:
@Hypfer this seems to work OK, but still doesn't solve the issue with "loading state" after wifi config.

```
root@rockrobo:~# cat /etc/init/valetudo.conf
#!upstart
description "Valetudo"

start on (started networking)
stop on runlevel [06]

oom score 1000
exec /usr/local/bin/valetudo
respawn
respawn limit 10 90
root@rockrobo:~#
```

Hypfer, 2019-04-28:
I don't think that that is easily solvable :/ Just reboot.

Your changed upstart config might cause firmware rollbacks

pmazanec, 2019-04-28:
reboot doesn't help, you need to start valetudo manually from ssh in ap mode, in order to get it run at all. why firmware rollbacks ? how about building a new fw with adjusted upstart script, would that help ?

Hypfer, 2019-04-28:
That should not happen with `start on net-device-up IFACE=wlan0` so there's definitely something wrong with the FW image.

Try building a new one with a completely new build folder

pmazanec, 2019-04-28:
I have just flashed it again with same results. please see my build script here:

https://gist.github.com/pmazanec/f6c3e39f750ee48987c7aefb4656767f

Hypfer, 2019-04-29:
Looks ok to me

imbaimbaimba, 2019-05-18:
My Valetudo eventually lived up after 2-3 days of loading. It worked fine, every day, since then, but now it just killed itself again :(

I might just have to eat my pride and download the Mi app and let them spy on me. 😂

loswochos, 2019-05-24:
Same here. Flashed the robi (S50) as described in the readme. Robi lost the wifi connection after 3-4 days.

The map is also not working.

pmazanec, 2019-06-15:
same happened to me, connection lost. never happened on the original firmware. rolling back ..

simonwood0609, 2020-01-10:
Any of you guys manage to find a permanent solution here?  Mine seems to stay alive for several days then disconnects from WiFi.

While you're here, other than completely resetting & re-flashing from scratch, is there another way to get it to reconnect to my WiFi while running Valetudo?

DerGute, 2020-01-13:
> Any of you guys manage to find a permanent solution here? Mine seems to stay alive for several days then disconnects from WiFi.
> 
> While you're here, other than completely resetting & re-flashing from scratch, is there another way to get it to reconnect to my WiFi while running Valetudo?
> EDIT: I think I've discovered how to re-connect it back to the WiFi without having to flash. Unfortunately, I reset it first, so now I have to re-flash. :(

Would you like to share your discovered way? I have the same problem. i dont want to reset my whole configuration. I really appreciate the Export of the live map with valetudo, but it is really necessary that it is reliable.

imbaimbaimba, 2020-01-13:
@simonwood0609 and @DerGute > Next official firmware from Roborock will enable room detection etc. on the S5. So I do not not see any reason to mess around with flashing firmware anymore. :-)

DerGute, 2020-01-13:
> @simonwood0609 and @DerGute > Next official firmware from Roborock will enable room detection etc. on the S5. So I do not not see any reason to mess around with flashing firmware anymore. :-)

Hey imbaimbaimba,
the advantage i see when messing around with flashing firmware is, that i can include the livemap to my smarthomemanagement. The independency from XIAOMI is another advantage.

imbaimbaimba, 2020-01-13:
> Hey imbaimbaimba,
> the advantage i see when messing around with flashing firmware is, that i can include the livemap to my smarthomemanagement. The independency from XIAOMI is another advantage.

That’s true, but I do not feel it is worth it. I don’t think the firmware resetting will get fixed as it has been going on this long. So I weighed it versus the frustrations of every time it goes dead, and I just reverted to original firmware. Actually, I don’t even have my vacuum on the network right now, I just lift it to wherever I need to clean, press the button and close the door. 

When the new firmware comes, I will connect it to the VLAN I have with other sketchy devices. I can then still connect it to HA etc if I want, but not even sure I will bother.

hillsalex, 2020-01-24:
I'm also experiencing it. @simonwood0609 - can you share your solution? This is super frustrating on my v1...

TerrorSource, 2020-02-11:
> @simonwood0609 and @DerGute > Next official firmware from Roborock will enable room detection etc. on the S5. So I do not not see any reason to mess around with flashing firmware anymore. :-)

Where's the source for this?

closed 2020-05-15 by Hypfer