Hypfer/Valetudo #81: Add Cleaning history page.

wbaars, 2019-01-20:
It would be nice to have a cleaning history page added to Valetudo.
Maybe show last weeks or last 7 cleanings.

stefanct, 2019-02-16:
What information would/should that include? Some ideas:

- Start timestamp
- End timestamp
- Duration
- Area cleaned
- Snapshot of the map
- Energy used

Bennedetto, 2019-02-17:
I just implemented most of the suggested functionality and created a Pull Request for the changes.
To keep the implementation efford as low as possible, I only implemented those items that are already provided by xiaomi. Thus the start-time, duration, area and possible errors are listed. 

The remaining ideas would require additional information that are currently not available.

wbaars, 2019-02-17:
Bennedetto, your implementation looks great!
Can't wait to test it out!

Bennedetto, 2019-02-17:
No need to wait for testing it. Just download the release 0.0.10b on https://github.com/Bennedetto/Valetudo/releases/tag/0.0.10b.
This contains all my changes that are currently published as PRs.

closed 2019-02-18 by Hypfer