Hypfer/Valetudo #83: Valetudo won't save settings

michel72, 2019-01-28:
Rooted a Roborock S50, firmware v3.3.9_001710 (downgraded to 001702)

Instaled valetudo, webuit works.

Changed the wifi settings, connection lost.

Then reset the wifi again, tried saving the sound settings, but also that fails. Seems to be an issue saving settings. Is does read the firmware version, token, etc.

Anyone?

michel72, 2019-01-28:
Right after settings the wifi, it does show up in my wifi network briefly, I can ping it, but no webuit and not ssh possible.

michel72, 2019-01-28:
Looks to be a more generic problem.

When setting the wifi using the descibed method on the DC website:

michel72, 2019-01-29:
Someone advised me to use https://github.com/LazyT/rrcc/ and that did work setting wifi :)

closed 2019-01-29 by michel72

Hypfer, 2019-01-29:
Interesting. Both tools call the same miio command. I wonder what might be different 🤔

michel72, 2019-01-29:
I've tied several times using mirobo, but it failed.

Hypfer, 2019-01-29:
Any special characters/spaces/etc. in SSID or password?

michel72, 2019-01-29:
yes, but it's the same ssid and pass I used with rrcc.

It's almost the same as:

ssid: kkmm
pass: KkMm.17.Nlt.15-mklsvdfers.Hnssfr

michel72, 2019-01-29:
I also tried ssid: tester and pass: 1234567890 to exclude the password character issue.

Hypfer, 2019-01-29:
And tester:1234567890 did not work with mirobo nor valetudo?

michel72, 2019-01-29:
No. I did not test this specific ssid with rrcc, since rrcc worked right away with my normal ssid.

michel72, 2019-01-29:
The only difference I can think of is that I used a different ssh key, since rrc required a private key wothoud password do I added one specifically for rrcc, but that should not be the issue either since I was able to connect via ssh when using the AP mode with roborock ssid.

Hypfer, 2019-01-29:
https://github.com/Hypfer/Valetudo/blob/master/miio/Vacuum.js#L300
https://github.com/LazyT/rrcc/blob/8b4800a999c248a393932a2161ae2c926ba3a91d/mainwindow.h#L111
https://github.com/rytilahti/python-miio/blob/f08b6153f4b68b4c1fea5cccaf3f6c94cdb92395/miio/device.py#L350

Okay so the only difference here is that rrcc uses null instead of 0?