Hypfer/Valetudo #84: Map path is wiped out every 15 mins

ideasman69, 2019-01-29:
Hi there,

Firstly - i LOVE this app, so thanks for putting in so much time making it 👍 

Secondly, I tried searching through for an existing issue but couldn't find one. Anyway - the _only_ complaint I have is that the map path wipes itself out every ~15 mins or so. I have ours hooked up to homeassistant and have the map png being shown on our 'vacuum page'. 

Our usual clean takes just over an hour to complete and once ~15 mins is hit, any path that it's taken is wiped out and the map starts fresh. Looking at the homeassistant log shows the same:

[image]

Just wondering if this is something easily fixed or whether it's possible to offload the data to be rendered elsewhere (if the vacuum's memory is being filled over that 15 mins). 

Thanks again

michel72, 2019-01-29:
I'm having the exact same issue, running firmware 3.3.9_001702.

michel72, 2019-01-30:
I flashed version 3.3.9_001712 today, and the issues is still present, but now seems to occur after ~25 minutes.

aluminium1, 2019-02-04:
Same issue for me after around ca 25 minutes on 3.3.9_001792

thomas-kuehn, 2019-02-07:
Same here with _001792.

ideasman69, 2019-02-11:
I've reverted back to stock firmware as we rely on the map too much. Will revisit down the track

legomind, 2019-02-12:
Just confirming the path resets on the map after about 30 minutes or so... The robot seems to still clean from room to room just fine though.

michel72, 2019-02-23:
The problem still persists with valetudo 0.2.0 on the S50 running 001792

michel72, 2019-02-23:
Just tested with 0.2.0 on 001730 which seems to be newer then 001792 and the problem still persists,

petabyteboy, 2019-02-24:
After having a quick look at how the code works, it seems like the robot regularly creates new logfiles, and only the latest logfile is parsed. It should be possible to read the map from the latest map file, but parse all logfiles for the robot's path and join them together. I'll try to work something out, that is also viable performance-wise (if the old files don't change anymore, we don't have to parse them again every time the map is reloaded).

michel72, 2019-02-24:
Sounds like a plan! :thumbsup:

Hypfer, 2019-02-24:
@petabyteboy See https://github.com/Hypfer/Valetudo/issues/86

14rf31, 2019-03-16:
HI all,
I enhanced the code, so that multiple log-files are parsed. Works for me now. 
[multiple_logs.diff.txt](https://github.com/Hypfer/Valetudo/files/2974150/multiple_logs.diff.txt)

aluminium1, 2019-03-16:
Pull requests?

14rf31, 2019-03-17:
My solution is not working under any condition. I have to investigate this and will come back to you.

closed 2019-04-06 by Hypfer