Hypfer/Valetudo #90: Refactor index.html

Hypfer, 2019-02-10:
Seriously. Just do it.
It hurts to see this abomination of a "WebApp".

legomind, 2019-02-10:
Good grief. If it hurts you that bad to use free software, then don't use it.
Problem solved.

legomind, 2019-02-10:
Lol you had me ready to comment about complaining about free software before I realized you were the repo owner.
Haha.

Hypfer, 2019-02-10:
I'm thinking about dropping Onsen UI.
It was easy to use the example and just change some elements, however splitting it up doesn't feel quite right.
Also, it's not really responsive but rather a mobile UI which gets very large on desktop systems and renders with a different CSS if you're on android.

This decision isn't final, though.

MeisterTR, 2019-02-10:
whats with the vue exaple of wolfspiritM [https://codesandbox.io/s/52y028o3qn](url) or maybe with vuetify (modern Material) i started a few weeks ago with vue. but i think im not good enoght to realize this... if i have some free time i will try..

Hypfer, 2019-02-10:
There's not just the example but rather a (seemingly nearly complete) branch in his fork:
https://github.com/WolfspiritM/Valetudo/tree/vue-integration

Poeschl, 2019-02-13:
Are they plans to continue on your fork @WolfspiritM ? Otherwise I would fork and take on the missing points.

closed 2020-05-01 by Hypfer