Hypfer/Valetudo #91: Multiple Maps

sirtet, 2019-02-11:
Questions about Idea for a new Feature:
I am trying to decide which robot t buy. 
As i understand, they all only support a single map, so are not comfortably usable for non-connected rooms Like different floors, right? 

Reading here, i wonder if it Would be possible to implement multi rooms by eg:
Let the robot record a map of every area, save it somewhere and add some Interface to swap maps whenever you carry it to the New area? Web UI or maybe even via a Hardware mod, Like a nfc sensor/bt beacon? 

Or, could you simply stitch the different maps together? I guess the robot Would happily just clean the area it can currently reach? 

Not being able to reach the charger base will always be an issue i guess. 
Do they send an SOS if they are unable to dock? 

How unrealistic am i?

cryptomilk, 2019-02-11:
See #86 and contribute :-)

Hypfer, 2019-02-11:
Since even Gen2 robots are currently priced around 320€, hacking the firmware so that it supports map swapping isn't reasonable at all. It's significantly cheaper to just buy as many robots as there are floors instead of spending weeks on this feature.

Even if you'd manage to get it to work it would still introduce lots and lots of new bugs since the roborock software is just not made for this use-case.
Therefore this is not and will not be supported by Valetudo

closed 2019-02-11 by Hypfer

sirtet, 2019-02-12:
> See #86 and contribute :-)

will try to if i get a robot, but i'll need tasks other than coding...

> ... It's significantly cheaper to just buy as many robots as there are floors instead of spending weeks on this feature.

Good point, but

- my old house has 3 floors and on 2nd floor 4 non-connected rooms, aka 6 robots...
- one might just not want a robot and dock in every room, for space, esthetics or whatever.
- one might want to save energy and physical resources, even if they are cheap to buy.

> Even if you'd manage to get it to work it would still introduce lots and lots of new bugs since the roborock software is just not made for this use-case.
> Therefore this is not and will not be supported by Valetudo

Fair enough. I just saw #69 also mentions the use of multiple floors, so i thought it might be common practice.

I don't know anything about vacuums yet, just trying to decide what to buy. 
But as i like hackability, i guess i'll go for a roborock s5 and see what happens.

treksler, 2019-03-06:
This was the only feature holding me back from Getting a Roborock over the Neato, which supports saving maps for up to 3 floors. Thank you for pointing me to the instructions in #69 
They look trivial enough.

I placed an order for the Roborock and will play with this when it arrives. Even if it won't be accepted into Valetudo, it shouldn't be too hard to make some kind of web UI to backup/save and switch/load maps.

If this works, I will probably get separate docks for my basement and for upstairs. It would be really neat to be able to trigger the map switching automatically when connected to the correct dock. I am assuming there is no way for the vac to know which dock it is connected to. is there onboard nfc or bluetooth or anything else that could read a tag of some sort? any other ideas?

roflcoopter, 2019-04-13:
> This was the only feature holding me back from Getting a Roborock over the Neato, which supports saving maps for up to 3 floors. Thank you for pointing me to the instructions in #69
> They look [...]

Do you have solution for the multiple floors problem? I am looking to do so myself otherwise

BoukeHaarsma23, 2020-01-02:
Just wanted to add something here. Apparently there is a beta with multi-room support floating around.

See [the announcement on facebook.](https://www.facebook.com/groups/1963580733664463/permalink/2865795586776302/)

If somebody has there hands on it. Maybe it is possible to use this in valetudo as well?