Hypfer/Valetudo #92: Map completely hang when robot runs, canvas experiment works

cryptomilk, 2019-02-11:
I'm running v001712 on a gen2 robot. When selecting map, it runs for a long time and then firefox pauses the script and asks If I want to wait or stop it. The canvas experiment from @matthiasharrer  works just fine. The map loads and I can see where the robot is. Maybe it is time to move :-)

Hypfer, 2019-02-11:
The reason why the other map performs better is that it doesn't try to fit the map to the screen.
Without that, the current drawing logic would be much faster

matthiasharrer, 2019-02-13:
Hi all :)

I think one problem with the "canvas-experiments" map (I wonder why I never changed this title ..) is that it does not really fit into to same spot as the other map usability wise.

# How would you imagine the map feature to be used?

How is the user supposed to interact with the embedded map?

@Hypfer pointed out some time, that the map for him was just a "gimmick". I am also thinking that its not that important in day-to-day interaction, but is really crucial in user experience for defining zones etc.  

My vision for the map would be to for example pop up as a modal dialog when defining zones, points, nogo zones or virtual walls. This is what I was trying to work towards with this https://github.com/matthiasharrer/Valetudo/tree/create-locations-from-outside.

# Prototype of /zone integrated into main UI

Nonetheless I replaced the 3rd party touch library for a simpler implementation where I can prevent events from being handled by OnsenUI. This would technically allow for integrating the map. See https://github.com/matthiasharrer/Valetudo/tree/integrate_map_into_onsen_ui.

# Concerning the Performance

Concerning the performance I actually don't think the **currently used** (not sure about the version using more efficient encoding on the API side) drawing procedure performs better. Just some simple measurements suggest that the /zone drawing is about 10x faster even when commenting out all rotating / scaling logic: https://github.com/matthiasharrer/Valetudo/tree/map-performance-comparison

leonardpitzu, 2019-02-14:
i think the map is useful in more scenarios. One would be to see where the robot got stuck and possibly to move it manually. Also it's good to know if certain areas were covered or not. As such i'd keep the map "visible". As for defining zones/virtual walls - yes, a good user experience is needed.

cryptomilk, 2019-02-15:
@Hypfer how do we move forward? I think we wait for your input ...

Hypfer, 2019-02-15:
I'd gladly replace the drawing logic, however the new one will need some logic which does the initial zoom.

As mentioned in #66, the Map render logic should be moved to a completely seperate project, which also means that it would be easy to use it in multiple places, so that we can keep the map as it is now to cover the use-cases @leonardpitzu mentioned, but also enable new ways of interacting with it like @matthiasharrer is doing.

Hypfer, 2019-02-19:
Fixed with 0.2.0

closed 2019-02-19 by Hypfer