Hypfer/Valetudo #95: Use saved cleaning zones in Home Assistant?

imbaimbaimba, 2019-02-16:
Hey guys,

When I save zones in Valetudo, can I then issue a clean of those from Home Assistant?
It would be great, as I have used the "add area to existing zone" a lot, so my zones are multiple coordinate sets. That seems necesarry for many of our zooms, as they are not rectangular.

TinusScheppers, 2019-04-26:
Is there a possibility to send zone coordinates or start in Valetudo configured zones using MQTT?
Or is it only possible using the 'legacy (platform: xiaomi_miio)' implementation in Home Assistant?

Hypfer, 2019-04-26:
It's currently WIP @TinusScheppers

TinusScheppers, 2019-04-26:
> 
> 
> It's currently WIP @TinusScheppers

That's great to hear, thanks for letting me know.

Keep up the good work!

closed 2019-06-04 by Hypfer