Hypfer/Valetudo #100: Port change

Anon0511, 2019-02-18:
Could make the port as a variable in the .conf (or a new .conf, doesn't matter) so the user can change the port?

Hypfer, 2019-02-18:
While that would be very easy to achieve, I can't help but wonder why anyone would want that.

Do you want to run more services?

https://github.com/Hypfer/Valetudo/blob/master/Valetudo.js#L18
It's already possible via a Environment Variable btw.

Anon0511, 2019-02-18:
Hey,

thank you!

I already expected this question. My main goal would be to make the service available from extern. Which basically at least would need SSL, so I can change the port to 8080, make the firewall restrictive and put a reverse proxy in front. However, yes, I had the idea to put some things via Docker there. 

FYI: https://www.roboter-forum.com/index.php?thread/33145-neuling-hat-keine-ahnung/

Anon0511, 2019-02-18:
Can't find it though. Am I just dumb on this one?

root@rockrobo:~# locate Valetudo.js
root@rockrobo:~# printenv
TERM=xterm
SHELL=/bin/bash
SSH_CLIENT=192.168.178.33 51535 22
UPDPKG_INFO_NAME=Update.pkg.inf
UPDPKG_INFO_PATH=/mnt/data/.temp
SSH_TTY=/dev/pts/0
LC_ALL=C
WIFI_ENABLE_NAME=wifi_enable.conf
WIFI_START_NAME=wifi_start.sh
USER=root
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
RR_RESERVE=/mnt/reserve
MAIL=/var/mail/root
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/rockrobo/rrlog:/opt/rockrobo/cleaner/bin:/opt/rockrobo/wlan:/opt/rockrobo/watchdog:/opt/rockrobo/miio:/opt/rockrobo/firmware
WIFI_CONF_NAME=wifi.conf
RR_DEFAULT=/mnt/default
PWD=/root
RR_ROOT=/opt/rockrobo
SHLVL=1
HOME=/root
NAV_CONFIG_PATH=/opt/rockrobo/cleaner/conf
LOGNAME=root
WIFI_START_PATH=/opt/rockrobo/wlan
RR_UDATA=/mnt/data
SSH_CONNECTION=192.168.178.33 51535 192.168.178.39 22
LESSOPEN=| /usr/bin/lesspipe %s
WIFI_ENABLE_PATH=/mnt/data/wlan
NAV_DYNAMIC_CONFIG_PATH=/mnt/data/rockrobo
DISPLAY=localhost:10.0
LESSCLOSE=/usr/bin/lesspipe %s %s
NAV_LOG_PATH=/mnt/data/rockrobo
WIFI_CONF_PATH=/mnt/data/miio
_=/usr/bin/printenv
root@rockrobo:~# printenv | grep 80
root@rockrobo:~# env
TERM=xterm
SHELL=/bin/bash
SSH_CLIENT=192.168.178.33 51535 22
UPDPKG_INFO_NAME=Update.pkg.inf
UPDPKG_INFO_PATH=/mnt/data/.temp
SSH_TTY=/dev/pts/0
LC_ALL=C
WIFI_ENABLE_NAME=wifi_enable.conf
WIFI_START_NAME=wifi_start.sh
USER=root
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
RR_RESERVE=/mnt/reserve
MAIL=/var/mail/root
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/rockrobo/rrlog:/opt/rockrobo/cleaner/bin:/opt/rockrobo/wlan:/opt/rockrobo/watchdog:/opt/rockrobo/miio:/opt/rockrobo/firmware
WIFI_CONF_NAME=wifi.conf
RR_DEFAULT=/mnt/default
PWD=/root
RR_ROOT=/opt/rockrobo
SHLVL=1
HOME=/root
NAV_CONFIG_PATH=/opt/rockrobo/cleaner/conf
LOGNAME=root
WIFI_START_PATH=/opt/rockrobo/wlan
RR_UDATA=/mnt/data
SSH_CONNECTION=192.168.178.33 51535 192.168.178.39 22
LESSOPEN=| /usr/bin/lesspipe %s
WIFI_ENABLE_PATH=/mnt/data/wlan
NAV_DYNAMIC_CONFIG_PATH=/mnt/data/rockrobo
DISPLAY=localhost:10.0
LESSCLOSE=/usr/bin/lesspipe %s %s
NAV_LOG_PATH=/mnt/data/rockrobo
WIFI_CONF_PATH=/mnt/data/miio
_=/usr/bin/env
root@rockrobo:~# env | grep 80
root@rockrobo:~#

Hypfer, 2019-02-18:
You need to edit the valetudo.conf in /etc/init/

https://github.com/Hypfer/Valetudo/blob/master/deployment/valetudo.conf
http://upstart.ubuntu.com/cookbook/#environment-variables

cryptomilk, 2019-02-18:
Why don't you just use a VPN, this is probably the most easy and secure solution.

Anon0511, 2019-02-19:
Simply because a VPN is not universal. This is rather just for fun anyway, as a SysAdmin/AppManager I like to try new things.
However if I don't get zone cleaning working today I will revert back to OFW anyway... sry but it is a must-have-basic function in my case.

cryptomilk, 2019-02-19:
See the currently open Pull Request?

Hypfer, 2019-02-19:
No need to apologize. Feel free to use whatever you like however you want :)

Anon0511, 2019-02-19:
Finally I got my first room "zoned". I guess I will open another issue about it (if it doesn't interfer with the Pull Request mentioned before).

Back to the port. My guess would be "env PORT=X"?

closed 2019-02-21 by Anon0511