Hypfer/Valetudo #102: Coordinates of the zones in map

MeisterTR, 2019-02-22:
would be nice to see the coordinates of the zone in the map

rassaei, 2019-02-22:
I back this a lot. Is there another way to get the coordinates of the map to create zones? I've read that version 0.1 showed you the coordinates. Why did you remove this? Maybe we can help fix this :)

Hypfer, 2019-02-22:
@rassaei The feature itself wasn't removed. The whole map component got swapped out for a new one which doesn't feature this yes.

matthiasharrer, 2019-02-23:
Hi all,

I added the coordinates to the /zone map again in #104.

What do you want to do with the displayed coordinates?
1) Do you want to use them to add them to the stored valetudo zones?
2) Or are you using them for other systems like home assistant etc.?

If its just for the first reason I would rather reopen #37 and close this issue as I think creating the stored zones using the map is the better option.

herrwusel, 2019-02-23:
Thanks @matthiasharrer . For me I use coordinates for 1. mostly.

imbaimbaimba, 2019-02-23:
@matthiasharrer I would like to find coordinates to use in home assistant. In the previous version I created a zone out of multiple rectangles. So I got something like this: 
`'[[953,-5035,-2160,333,2],[-2160,-3720,-6441,333,2],[-2160,-5035,-2845,-3720,2]]'`
I would like to be able to get that again after drawing a couple of rectangles.

rassaei, 2019-02-26:
@matthiasharrer Thanks a lot!
Primarily I want to use them in Home Assistant, but still have to get everything up and running.
But as @imbaimbaimba said, saving zones would be awesome too, as afaik you could combine zones and create more than a rectangle with that feature?

TinusScheppers, 2019-05-03:
I'm trying to find the possibility to save drawn zones from the map, but this option does not seem to be there (anymore)?

jowlo, 2019-05-03:
Actually, it is not there _yet_, however, it has Bern merged into `master` and will be in the next release, see #157.

Am 3. Mai 2019 10:42:29 MESZ schrieb Tinus Scheppers <notifications@github.com>:
>I'm trying to find the possibility to save drawn zones from the map,
>but this option does not seem to be there (anymore)?

jowlo, 2019-05-03:
Actually, it is not there **yet**, however, it has been merged into `master` and will be in the next release, see #157.

TinusScheppers, 2019-05-03:
Looks really neat, thx!

jowlo, 2019-05-03:
All thanks redirected to @matthiasharrer ;)

imbaimbaimba, 2019-05-03:
I am still a bit confused about how to get coordinates that can be used outside of Valetudo. The coordinates found by the /zone tool never did work for me in Athom Homey or Home Assistant - as they need the actual Xiaomi coordinates.

I read that people get these in various ways, like say with Flolevac, but since my vacuum was never used with the mi app, I cannot use that. 

I also read somewhere that double clicking would give the real coordinates, but I haven’t had any success with that either.

Any help on retrieving the real coordinates from Valetudo?

llonca13, 2019-05-06:
Will it be possible with this new update to get coordinates from the map as @imbaimbaimba said ('[[953,-5035,-2160,333,2],[-2160,-3720,-6441,333,2],[-2160,-5035,-2845,-3720,2]]') in this format?
To be used with Home Assistant?  Thanks

simonwood0609, 2020-01-10:
How can I get coordinates out of Valetudo?

TheLastProject, 2020-05-03:
@Hypfer Would you accept a pull request that exposes the zones in the /api/map/latest API call and the MQTT output? I believe I remember there were changes being made to the map API so I prefer to ask to prevent getting in the way of that.

Hypfer, 2020-05-04:
@TheLastProject maybe a differend endpoint/topic?

aebgit, 2020-08-04:
Is the coordinate-system used by the map-editor in Valetudo different from other solutions that employ MIIO library (eg. [homebridge-xiaomi-roborock-vacuum](https://github.com/homebridge-xiaomi-roborock-vacuum/homebridge-xiaomi-roborock-vacuum))?
I'm asking because I have no luck using the coordinates provided via `/api/get_config`.

Hypfer, 2020-08-04:
@aebgit yes

aebgit, 2020-08-04:
Is there a way to convert or look them up on OS-level on the vacuum, or is this a dead-end?

zvldz, 2020-08-04:
Convert from Valetudo (condition - x1 < x2, y1 < y2)
 x1 = v_x1 
 y1 = 51200 - v_y1
 x2 = v_x2
 y2 = 51200 - v_y2

Hypfer, 2020-08-04:
@zvldz Thats most likely incorrect with never valetudo version which is also the reason for why it's unsupported to do things with coordinates outside of valetudo.

@aebgit Usually the thing you're trying to do is something you shouldn't do. Otherwise there's no real need for coordinates outside of Valetudo

zvldz, 2020-08-04:
> Thats most likely incorrect with never valetudo version which is also the reason for why it's unsupported to do things with coordinates outside of valetudo.

Well, then it seems like:
x1 = v_x1*10
y1 = 51200-v_y1*10
x2 = v_x2*10
y2 = 51200-v_y2*10

for roborock

aebgit, 2020-08-04:
@zvldz Thank you, I'll try!

Hypfer, 2020-08-04:
@zvldz The problem with providing these formulars is that they can and will break at any time which then causes further issues down the road

Please don't do that

zvldz, 2020-08-04:
> The problem with providing these formulars is that they can and will break at any time which then causes further issues down the road

I agree. 
It is possible to tell that the current formula can be found in file Roborock.js (for roborock).

abmantis, 2020-08-11:
@Hypfer why does the formula changes and why not use the raw values from miio?

Hypfer, 2020-08-11:
@abmantis Because the goal of valetudo is to be a vendor-agnostic abstraction of all specifics so that you can have one unified format for different robots, vendors etc

abmantis, 2020-08-11:
@Hypfer makes sense. Any recommendation on how to get and set no-go zones from an external service (like Home Assistant)? My plan is to have some toggles to remove/add some no-go zones or barriers.

Hypfer, 2020-08-12:
@abmantis I'd strongly advice against trusting the robot to keep the map in a persistent state.

It can change at any time which then means that your offsets are completely wrong and those walls will be all over the place.
This feature needs a real human looking at the visual representation of the map to make sure that everything is correct.

What are you trying to achieve exactly? Why would there be dynamic no-go zones with fixed coordinates?

abmantis, 2020-08-12:
@Hypfer I've had my S50 for about a year now and it has been consistent. The zones have never gone wrong (I have about 5) and the map was always persistent and consistent.

My use case is the following: There are some things on the floor that I want the device to avoid on the daily cleanup. Weekly, I pick up some of those things from the floor so that the vacuum cleans there too.
So, the daily cleanup has 7 no-go zones, while the weekly has only 3, for example. Having a way to toggle this from HA would be much better than removing them manually.

Hypfer, 2020-08-12:
@abmantis
Would it be possible to use the segment cleaning feature for that and simply segment in a way that you can exclude said zones?
Of course that would also require room editing support in Valetudo which isn't there yet 👀

abmantis, 2020-08-12:
@Hypfer that would be a bit akward to do, because most of the no-go zones are small (like a body scale) and in the middle of another room.

What about providing the current zones in mqtt, and allowing setting them trough mqtt? 
Another (better IMO) option is to be able to define "groups" of no-go zones in valetudo, and be able to select the group trough mqtt. Similar to the current multi-map feature, but for no-go zones.

Hypfer, 2020-08-12:
@abmantis 
There's no multi-map feature in Valetudo so you're probably referring to Valetudo RE.
You're use-case might indeed be better suited for that project, because I really can't implement something that works well 95% of the time but may cause catastrophic results for the other 5% of cases :(

abmantis, 2020-08-12:
@Hypfer oh, sorry. I though it was also available in Valetudo too. Yeah, I completely understand that.

closed 2021-03-23 by Hypfer