Hypfer/Valetudo #106: draw robot and charger on map

petabyteboy, 2019-02-24:
How can I make Valetudo draw the robot and the charger on the map in the web interface?

Hypfer, 2019-02-24:
It did that at some point. Guess it's missing in the new 0.2.0 Map.

@matthiasharrer ?

petabyteboy, 2019-02-24:
I will try to add it back.

matthiasharrer, 2019-02-24:
You can have a look at https://github.com/matthiasharrer/Valetudo/tree/add-robot-drawing?files=1

It seemed to work but the robots orientation was incorrect. Would be nice if you can figure it out:)

petabyteboy, 2019-02-24:
But it's exactly the code that was used before..? That's weird.

matthiasharrer, 2019-02-24:
Yup that's what confused me too.
Did you try it with your robot? Is it working for you?

petabyteboy, 2019-02-24:
Nope, the angle is very weird.

matthiasharrer, 2019-02-24:
Ah I think the problem is the flipping of the map. I'm not exactly sure but depending on where the map is read from the api returns a flag that indicates a flipped map. I think we have to account for that with the robots angle too.

petabyteboy, 2019-02-24:
I mean most of the time it looks okay, but just because of how this code works it will always look a bit weird in some cases.
[image]
[image]
[image]
[image]
You can see that the first two look fine, but the last two look off somehow.
Actually it makes a lot of sense that these angles have not been adjusted for a flipped map.

petabyteboy, 2019-02-24:
I don't understand this: The whole path should have been flipped to account for the map being flipped. So did the old code somehow flip the y axis, so it is flipped twice now? @Hypfer

petabyteboy, 2019-02-24:
Nevermind, the way this the path is used the transform is not being applied.

matthiasharrer, 2019-02-24:
I think the flipping is applied when the meter based path coordinates are transformed to the 1024*1024 px map coordinates. For calculating the angle I used the original path coordinates .. so they miss the flipping.

petabyteboy, 2019-02-24:
Yes, that's how it is. 
I am having a few other problems where the charger is shown in the wrong place, is that a known issue?

matthiasharrer, 2019-02-24:
You mean a little bit or completely off? We discussed it in #72 but I never really looked into it.

petabyteboy, 2019-02-24:
It was completely off. I stopped the robot in the middle of a run and after that it thought it was at its charging point and it couldn't find the charger when I told it to go home. So I guess not-our-bug.

Hypfer, 2019-02-24:
#107

closed 2019-02-24 by Hypfer