Hypfer/Valetudo #114: Feature-Request: Play sound via api

klein0r, 2019-03-07:
Would be great to play custom wav files using the API.
This is currently possible by using the following command

```
/opt/rockrobo/firmware/uart_test -e /mnt/data/rockrobo/sounds/error1.wav
```
(maybe you know another solution)

I would create a pull-request. But I have no idea where to search for. I've endet up in "sendMessage" with parameter "test_sound_volume" on a udp socket?! 😢

Bennedetto, 2019-03-07:
What is the use case for this? Do you want to play the available soundfiles? Or uploading custom sounds to play them? Or generate new ones, for e.g. using pico2wave as tts with wav output - and uart_test as sound output? 

In the past I had installed pico2wave (packages "libttspico-utils" and maybe "sox" are required)
on my Roborock S50, thus it is able to speak custom commands - but currently I do not see any usefull application for that..

An example workflow could be:
/usr/bin/pico2wave -w /tmp/dynamic_speech.wav -l en-US "Hello World! This is my amazing robotic voice!"
/opt/rockrobo/firmware/uart_test -e /tmp/dynamic_speech.wav

Maybe smart home systems could send messages to the robot as a audio sink.. (like sending text to robot, let the robot synthesize to sound, let the robot "speak" the synthesized text)..

I would love to hear your ideas!

klein0r, 2019-03-08:
A lot of people are searching for audio feedback in their smart homes. There are a lot of different solutions for that. e.g. SONOS.

For example: When you enter the room, you want to get a list of open tasks / whats happening today. Or someone rings on the front door, you could also play a sound over your vaccum without a seperate speaker in that room.

Obviously I want to play my own sound files. Maybe valetudo could offer an upload mechanism on the web UI to make that process easier. What I need for that task, is an API which can be used to trigger the playback.

At the moment I could use ssh to start the playback. But that's not the nicest solution 😄

thundergreen, 2019-04-12:
I'd be also interested in this

TheMicroDevil, 2019-04-25:
If you only want to play audio files and have some remote control over LAN, this could be a solution: 
[https://github.com/cmus/cmus/wiki/remote-control](https://github.com/cmus/cmus/wiki/remote-control)
cmus can connect to servers, so you could use playlists or streams without local audio files on device.
Or maybe mpd?

By using valetudo WebApi (/api/current_status) you can do many crazy things and trigger different audio files/playlists/streams ;)

`{"msg_ver":8,"state":8,"battery":100,"clean_time":40,"clean_area":705000,"error_code":0,"map_present":1,"in_cleaning":0,"fan_power":38,"dnd_enabled":0,"human_state":"Charging"}`

As some idea sharing for custom audio files and crazy ideas....
I have done this, on device, in bash.
First (dirty buggy) POC-release can be found here (german): [https://www.roboter-forum.com/index.php?thread/33070-miplay-v0-7-beim-saugen-musik-inkl-lautst%C3%A4rkeregelung-abspielen/&postID=411231&highlight=miplay#post411231](https://www.roboter-forum.com/index.php?thread/33070-miplay-v0-7-beim-saugen-musik-inkl-lautst%C3%A4rkeregelung-abspielen/&postID=411231&highlight=miplay#post411231)
- cmus as audioplayer
- Playing music while a job is running
- Volume change based on suction power in real time
- stop music when docked
- pause music when suction is paused (for example triggered from call monitor for FritzBox)
- Event based sound files (custom error message/job finished)

But playing files will block the sound device and you will not hear robot messages anymore.
So i'm trying some ideas to mix music + standard/custom audio messages.

Grygon, 2019-10-27:
I'd love this, would be a cool feature to implement that goes beyond the basic functions of a non-valetudo vacuum. 

In particular, I was inspired to try replicating this because of [dj roomba](https://www.youtube.com/watch?v=pXhsUPtsiLU), and while I can do that over CLI it'd be great to have a web interface instead.

scasic, 2020-01-13:
When I try to play wav file using uart_test I get an error:
```
/opt/rockrobo/firmware/uart_test -e /mnt/data/rockrobo/sounds/error10.wav
main:589:FIXME Do not response signal
```

citronalco, 2020-03-14:
> But playing files will block the sound device and you will not hear robot messages anymore.
> So i'm trying some ideas to mix music + standard/custom audio messages.

To avoid blocking the audio device use ALSA's dmixer plugin like described here: https://blog.geierb.de/musik-auf-dustcloud-befreiten-xiaomi-staubsauger/
What's still missing is downmix to mono and lto max. 22 kHz

thundergreen, 2020-03-14:
I just tried the hint from @citronalco  and my robot plays musik here in my living room :) Of course the robot messages don't work but thisis not really bad.. musik is much nicer :)

citronalco, 2020-03-14:
> I just tried the hint from @citronalco and my robot plays musik here in my living room :) Of course the robot messages don't work but thisis not really bad.. musik is much nicer :)

Have you rebooted the robot?
Music and robot messages are supposed to work at the same time.

thundergreen, 2020-03-14:
it is rooted and negative.. robot does not talk anymore

citronalco, 2020-03-14:
> it is rooted and negative.. robot does not talk anymore
> but i can finally listen to my favourite radio chanel :)

Strange. My robot can do both at the same time after creating asound.conf & rebooting.

thundergreen, 2020-03-14:
a! i did not reboot

Hypfer, 2020-05-15:
I'd suggest throwing another piece of software on the robot which does that

it's out-of-scope for valetudo

closed 2020-05-15 by Hypfer

duhow, 2020-10-31:
I was able to compile and run **mpd 0.19.21** along with **upmpdcli v1.1.4** - **libupnpp v0.14.1**, now I turned my vacuum Roborock S6 in a UPnP media renderer that can play music and radio while it cleans the house :joy:

citronalco, 2020-10-31:
@duhow Can you upload the binaries somewhere?

gmykhailiuta, 2021-04-14:
I was able to make text-to-speech as well as Alsa's `aplay` available on Roborock S5 (v3.5.8_002034 Valetudo) by combining two things:
1. Files from https://github.com/porech/roborock-oucher/raw/d93fe4e252d25c40dfbcb146c31e0f2e49fe5623/oucher_deps.tgz (was part of roborock-oucher project). But it lacked `aplay` (contains script which relies on `play` - absent on my unit).
2. Getting the armhf (arm 32 bit) `alsa-utils` package for Ubuntu Trusty (which my unit's firmware is based on) here http://ports.ubuntu.com/ubuntu-ports/pool/main/a/alsa-utils/alsa-utils_1.0.27.2-1ubuntu2_armhf.deb and extracting it to root of the firmware:
```
wget http://ports.ubuntu.com/ubuntu-ports/pool/main/a/alsa-utils/alsa-utils_1.0.27.2-1ubuntu2_armhf.deb
ar xv alsa-utils_1.0.27.2-1ubuntu2_armhf.deb
tar xf data.tar.xz
scp usr/bin/aplay root@roborock:/usr/bin/
```
So now it's possible to play wav files with `aplay file.wav`, as well as make robot say any text with `espeak -v engilsh "Hello world!"`.
Most likely, it's possible to "install" any Ubuntu package in that way. Tried to install mpd, but:
1. Ubuntu's mpd package was built with a ton of dependencies (avahi, etc.), which makes it a nightmare to extract and install all of them manually. So it might be worth automating it somehow, e.g. by installing apt. But:
2. There's not much space on the root partition (although it's possible to use /data, or modify partitions).

I found that `aplay` and `espeak` is more than enough for me, and `mpd` just doesn't worth the effort.

definitio, 2021-04-15:
There is a (custom component)[https://github.com/definitio/ha-sox] for home assistant. [This firmware](https://github.com/zvldz/vacuum) is required.

thundergreen, 2021-04-15:
takee this linke :) 

https://gitea.ddns.net/EmEvTh/ha-sox