Hypfer/Valetudo #115: Not cleaning the whole room anymore?

UnbekanntesPferd, 2019-03-11:
After rooting and installing valetudo on my S50, I'm facing the problem that not every time the whole room is cleaned. I'm pretty sure this can't be connected to valetudo as it's not touching the "cleaning logic" itself, but I'm curious if anyone else is noticing similar issues...?
[image]

matthiasharrer, 2019-03-11:
Hi,

there are known issues that the path is not drawn completely. (#84)
Are you sure the room is actually not cleaned? Or could it be that it is simply not shown?

UnbekanntesPferd, 2019-03-11:
Hi!
That would explain the semicircles on the top of the above map - they look pretty strange.
I will investigate that a bit further and post an update here.
Thanks!

matthiasharrer, 2019-04-05:
@UnbekanntesPferd did you figure out if it's the same issue as #84 ?

closed 2019-04-06 by Hypfer

UnbekanntesPferd, 2019-04-08:
Couldn't figure this out, as I had not been in the room while this happened so far.
But I think closing is ok. I will check again with v0.3.

cryptomilk, 2019-04-08:
The route is chosen by the robot, Valtetudo has no influence. Routing seems to change with every new firmware version, it seems there isn't the perfect solution found yet