Hypfer/Valetudo #116: Feature request: notifications

dh-harald, 2019-03-12:
I like this project, it's getting better and better... I've only one feature missing, that's the notifications.
I think, you should add somehow a configurable rest api, for sending messages...
I could use Mattermost or Slack or similar, or (I think, it's not sure), we could create HA endpoints to send whatever message provider, that we could use in HA. With help of this applications, we could send push notifications easier.
Or you can implement the push notifications natively, but in this case, you should handle android and ios differently

Hypfer, 2019-03-12:
If you're using HA, I guess the best solution would be to implement #43 and then do notifications using automations in HA on state changes

matthiasharrer, 2019-03-12:
If you know your token it is also quite easy to hook the robot up to HA directly (additionally to Valetudo) https://www.home-assistant.io/components/vacuum.xiaomi_miio/

Hypfer, 2019-03-22:
MQTT Support is done and available in 0.2.1

closed 2019-03-22 by Hypfer