Hypfer/Valetudo #119: Feature request: Using Gamepad for Manual Control

atomphil, 2019-03-22:
It would be fun to control the robot using a gamepad connected to the computer. Maybe it can be implemented using the HTML5 gamepad API.

bademux, 2019-03-25:
and stats, who cleans the room faster :D

timbru31, 2019-04-12:
I've made a POC and it works pretty well so far. Using nothing more than the Gamepad API and 50 lines of JS. 🎉 
(Not really ready for a PR yet)

timbru31, 2019-04-14:
@Hypfer - is there any documentation about the min and max values for speed, angle and duration? Haven’t found any so far.
(Seems velocity is from -0.3 to 0.3; angle from -1.5 to 1.5?)

Hypfer, 2019-04-14:
@timbru31 https://github.com/marcelrv/XiaomiRobotVacuumProtocol/blob/master/remote_control.md#command-2

There's this project which tried to document stuff but apart from that not much

timbru31, 2019-04-14:
Thanks! Helps a bit, but was nearly what I’ve got from looking at the network requests the current remote control sends. Interesting that Xiaomi didn’t make the speed from -1 to 1, but they’ve probably had a reason for it^^

timbru31, 2019-04-20:
Am I able to determine if I “crashed” into something from the current robo state? This could be used for a rumble feature ;)

timbru31, 2019-06-25:
Small heads up, I’m still working on this. Currently I’ve decided to make this a small extra page without the whole Onsen UI thingy. (See #90, too)

However, iOS 13 works fine with my DualShock gamepad 🎉

Hypfer, 2020-05-01:
While its cool, I think this is out of scope

closed 2020-05-01 by Hypfer