Hypfer/Valetudo #120: Device Info not showing 0.2.1

nightcat91, 2019-03-22:
Hi,

first of all, thank you for this great piece of software!!

I just updated my Gen2 to version 0.2.0 using this guide: https://github.com/dgiese/dustcloud/wiki/Cloud-Free-Firmware-Image-With-Valetudo
which is very good by the way!!

After I was done I saw the release of verison 0.2.1 and I updated the valetudo binary (grabbed the one from the release and uploaded it to the robot).
I did a "service valetudo stop" as suggested, changed the file and fixed permissions and started valetudo again.

But now when I connect via a browser everything seems fine and working apart from the Info view in the settings page (and also the cleaning history).

It gives a small message about a connection error.
See attached screenshot

If I go back to the 0.2.0 binary, everything is back and working...
I am using firmware version 3.3.9 and firmware build 1748

Let me know if any more info is needed or if I can help test anything

[image]

oechslein, 2019-03-22:
Same here, here is the log:
```
Loading configuration file: /mnt/data/valetudo/config.json
Webserver running on port 80
pkg/prelude/bootstrap.js:262
  if (error) throw error;
             ^

Error: File '/**/Valetudo/lib/package.json' was not included into executable at compilation stage. Please recompile adding it as asset or script.
    at error_ENOENT (pkg/prelude/bootstrap.js:422:17)
    at readFileFromSnapshot (pkg/prelude/bootstrap.js:650:29)
    at Object.fs.readFileSync (pkg/prelude/bootstrap.js:693:18)
    at /snapshot/Valetudo/lib/webserver/WebServer.js:78:45
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:442:3)
```

Hypfer, 2019-03-22:
I've replaced the 0.2.1 release binary with one built from the current master. Just redownload and it should be fixed.

 Sorry for the inconvenience

closed 2019-03-22 by Hypfer

oechslein, 2019-03-22:
Works now! Thanks for your quick help and in general for this master piece.

nightcat91, 2019-03-23:
Thanks for the quick fix!!