Hypfer/Valetudo #121: map location 0.2.1, API calls

nightcat91, 2019-03-23:
using the latest (fixed) v.0.2.1 I cannot get the location of the map png anymore from 
`YOUR.VACUUM.ROBOT.IP/api/remote/map?drawRobot=false&drawCharger=true&scale=5&border=3&doCropping=true&drawPath=true`
I dont know if this had to do with the update, but if I remember correctly it worked before updating to 0.2.1

Now if I enter the above URL in a browser I get this:
`TypeError: Cannot read property 'get' of undefined
    at /snapshot/Valetudo/lib/webserver/WebServer.js:802:32
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/snapshot/Valetudo/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/snapshot/Valetudo/node_modules/express/lib/router/index.js:335:12)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/index.js:275:10)
    at jsonParser (/snapshot/Valetudo/node_modules/body-parser/lib/types/json.js:110:7)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)`

I would still like to be able to get the map png since I am using this as a generic camera in HomeAssistant.

I have read about the MQTT addition in v.0.2.1 but I am not sure if this can help with the map....
See my other issue about the new mqtt feature here: (tbd)

Hypfer, 2019-03-23:
Ah yes. A stupid error due to me loosing track of things in that huge webserver.js.

Again, I've built a new binary with the latest master and replaced it in the release.
Redownload and it should be fixed.

I'd like to phase out the remote API but as long as there is no MQTT alternative its the only way to get the map I guess.

closed 2019-03-23 by Hypfer

nightcat91, 2019-03-23:
Thanks for the quick fix, its working now!