Hypfer/Valetudo #122: MQTT 0.2.1

nightcat91, 2019-03-23:
using the latest (fixed) v.0.2.1
I think MQTT is a great addition!!

But I am not sure I understood the release notes correct...
I should be going into the robot to edit the `mnt/data/valetudo/config.json` file and add this:
`"mqtt": {
	"enabled": false,
	"identifier": "rockrobo",
	"broker_url": "mqtt://foobar.example"
}`
But why does it say "enabled: false". Is that correct??

And I would need to be able to enter a user and a password, since my mqtt server only allows client with authentication. 

And the last question would be about the map... 
https://github.com/Hypfer/Valetudo/issues/75#issuecomment-451731250 here you said that the API should not be used too much since MQTT should be used. I think MQTT is great, but at least for HomeAssistant (which you linked in the release notes), there is no way to get the map over mqtt...
Also have a look at my other issue about getting the map location to grab the map:
https://github.com/Hypfer/Valetudo/issues/121

Hypfer, 2019-03-23:
Yes you will have to set `enabled` to `true` and add user & password like this: `mqtt://user:password@host`

I'm unsure how to do it so there's no mqtt map support yet but it should follow at some point since I'd too like to see it.

closed 2019-03-23 by Hypfer

nightcat91, 2019-03-23:
Thanks a lot.
I think the only way to replace the API call to generate the map would be to replace that with some mqtt commands that define how the map should be generated and an mqtt answer with the location and name of the new png map

Hypfer, 2019-03-23:
It's possible to just send the whole map image via mqtt.

https://www.home-assistant.io/components/camera.mqtt/

The mqtt camera component supports this.
I guess that could be a solution 🤔

nightcat91, 2019-03-23:
Thats a very good idea, and it could also be used to change the update timing of the map. So when the robot is charging or docked it can reduce the update frequency of the map and while its running it can increase it.
Because of this:
`Every time a message under the topic in the configuration is received, the image displayed in Home Assistant will also be updated.`

I have some problems getting the mqtt connection to work...
I edited the config file in two different ways, stop and restart valetudo but I dont get the web interface after it, only `ERR_CONNECTION_REFUSED` in the browser.

Where is a log I can have a look at and maybe find something?

I tried this version first:

Hypfer, 2019-03-23:
[image]

Looks like just piping pngs into a mqtt topic seems to work. Thats nice

Regarding your issues:
Your configuration is invalid.

```
{
	"spots": [],
	"areas": [],
	"mqtt": {
		"enabled": true,
		"identifier": "roborock",
		"broker_url": "mqtt://roborock:roborockmqtt@10.11.11.1"
	}
}
```
This should work

nightcat91, 2019-03-23:
Thanks for correcting my syntax.... i didnt think two of those brackets "{ }" inside each other would work.

Now I get the web ui back but its not finishing loading...
[image]

The mqtt server gets a connection and the robot also send its initial config message `homeassistant/vacuum/valetudo_roborock/config` which gets picked up by HomeAssistant.
But the robot also does not send any additional messages about its state or about a change in fan speed or anything else.

Hypfer, 2019-03-23:
If valetudo was running for a long time the message IDs may get out of sync which results in large amounts of retries.

Try rebooting the robot. That resets all counters back to 0

nightcat91, 2019-03-23:
Restarting fixed the webui and also the MQTT.
I am now able to controll the robot via mqtt, thats great!!

Is it possible to include the mop mode via the fan speed selector?
What I did when I was controlling the robot via its token locally (with valetudo installed), I send fan speed 105 to the robot, which is mop mode.

```  ```

    - service: vacuum.set_fan_speed
      entity_id: vacuum.roborock
      data:
        fan_speed: "105"

These are the fan speeds in numbers:
Fan Speeds: 101=Quiet, 102=Balanced, 103=Turbo, 100(104)?=Max, 105=mop

nightcat91, 2019-03-25:
I need to restart the robot about once per day otherwise the above problem with the webui comes back.
Is there a fix for that maybe?

Hypfer, 2019-03-25:
Check `/mnt/data/rockrobo/rrlog/miio.log` to see if you can find out more about that problem

nightcat91, 2019-03-26:
Here is what I right now in the log:

> [20190326 02:26:23] [INFO] Set data dir to: /mnt/data/miio/
[20190326 02:26:23] [INFO] Set log level to: 2
[20190326 02:26:23] [INFO] Set TCP host to "ott.io.mi.com"
[20190326 02:26:23] [INFO] Set UDP host to "ot.io.mi.com"
[20190326 02:26:23] [INFO] OT smart conn sockfd: 8
[20190326 02:26:23] [INFO] OT agent listen fd: 10
[20190326 02:26:23] [INFO] BT conn listen fd: 11
[20190326 02:26:23] [INFO] Timer fd: 12, interval: 5000 ms
[20190326 02:26:23] [INFO] Sync timer fd: 13, interval: 0 ms
[20190326 02:26:23] [INFO] ACK timer fd: 14, timeout: 0 ms
[20190326 02:26:23] [INFO] pollfds[0]: 8
[20190326 02:26:23] [INFO] pollfds[1]: 10
[20190326 02:26:23] [INFO] pollfds[2]: 11
[20190326 02:26:23] [INFO] pollfds[3]: 12
[20190326 02:26:23] [INFO] pollfds[4]: 13
[20190326 02:26:23] [INFO] pollfds[5]: 14
[20190326 02:26:28] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:28] [WARNING] No helper script?
{"method":"_internal.helper_ready"}
[20190326 02:26:33] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:33] [WARNING] No helper script?
[20190326 02:26:38] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:38] [WARNING] No helper script?
[20190326 02:26:43] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:43] [WARNING] No helper script?
[20190326 02:26:48] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:48] [WARNING] No helper script?
[20190326 02:26:53] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:53] [WARNING] No helper script?
[20190326 02:26:58] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:26:58] [WARNING] No helper script?
[20190326 02:27:03] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:03] [WARNING] No helper script?
[20190326 02:27:08] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:08] [WARNING] No helper script?
[20190326 02:27:13] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:13] [WARNING] No helper script?
[20190326 02:27:18] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:18] [WARNING] No helper script?
[20190326 02:27:23] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:23] [WARNING] No helper script?
[20190326 02:27:28] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:28] [WARNING] No helper script?
[20190326 02:27:33] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:33] [WARNING] No helper script?
Connect to server error: 127.0.0.1:54322
[20190326 02:27:38] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:38] [WARNING] No helper script?
[20190326 02:27:43] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:43] [WARNING] No helper script?
[20190326 02:27:48] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:48] [WARNING] No helper script?
[20190326 02:27:53] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:53] [WARNING] No helper script?
[20190326 02:27:58] [WARNING] cmd_internal_request_dinfo:1124, ot_agent_internalfd: -1
[20190326 02:27:58] [WARNING] No helper script?

after this it just goes on like the last few lines, repeating about every 5-10 sec.

I will do a restart and see if I can find anything else in the log

Hypfer, 2019-03-27:
What Firmware version are you using?

nightcat91, 2019-03-28:
I am using 0.2.1 (the second updated version that you uploaded for that release), the one after you made this comment:
https://github.com/Hypfer/Valetudo/issues/121#issuecomment-475847535

Hypfer, 2019-03-28:
And the Robot FW?

nightcat91, 2019-03-28:
Gen2: v11_001748