Hypfer/Valetudo #126: MQTT over tls?

dugite-code, 2019-03-26:
I would like to use MQTT over tls looking at the [Configuration.js](https://github.com/Hypfer/Valetudo/blob/master/lib/Configuration.js#L13) provided no hints as to if this is possible

dugite-code, 2019-03-26:
Ok I think I figured it out just using the [protocol](https://github.com/mqttjs/MQTT.js/issues/437#issuecomment-233924282) `mqtts` and adding the port on the end. Luckily I had letsencrypt set-up as I wasn't able to use the self-signed certificate I had set-up.

```
"mqtt" : {
            enabled: true,
            broker_url: "mqtts://user:pass@foobar.example:8884",
            }
```

closed 2019-03-26 by dugite-code

timbru31, 2019-04-06:
Hi @dugite-code,

I have to disagree with your comment
> [...] there isn't a quick way to deal with self signed certs without negating the whole point.

I've successfully managed to connect to my MQTT broker (mosquitto) which is using a self signed cert. :)
The TLS lib of Node.js accepts an array of custom CA's: https://nodejs.org/api/tls.html#tls_tls_connect_options_callback

With a little change in the `MqttClient.js` I was able to configure the CA path in the Valetudo options.

@Hypfer are you interested in a PR that adds the CA path config option?

Hypfer, 2019-04-06:
@timbru31 sure if its optional and doesn't cause any breakage

dugite-code, 2019-04-09:
@timbru31 Neat! I couldn't find any info on the self signed CA. I guess the issue was I was looking for it in the MQTT library docs instead of the nodejs docs

timbru31, 2019-04-26:
@dugite-code I've finally created the PR which enables this feature :)

PR #188 (2019-04-26): feat: add MQTT over TLS (mqtts) support
Added an option to specify the certificate authority (CA) file path to enable mqtts support. One has to copy e.g. the `mosquitto.crt` file from the MQTT broker to the robo. This is working because the mqtt npm package uses Node.js TLS lib which can use a custom CA list: https://nodejs.org/api/tls.html#tls_tls_createsecurecontext_options

Example would be:

`    "caPath": "/mnt/data/valetudo/mosquitto.crt"`

This fixes #126. I've successfully used this setup for some weeks.