Hypfer/Valetudo #127: A way to show MAP tab on default ?

thundergreen, 2019-03-28:
Dear all involved. It is just amazing how this project is going so far :) THUMBS UP!

As i use valetudo webinterface for my Dashboard. I was wondering if it would be possible or HOW it would be possible to show the MAP tab as default when valetudo webinterface is loading.

Just to give you an idea:

I implemented this webinterface in my Dashboard looking like this, but each time I call this dashboard it loads the main view which is not very useful for me.

[image]

I would rather like to have this view as fedault:

[image]

Further more the map seems to be aligned to the left. is it possible having the map centered?

Thanks for any hints and help

:) Cheers!

leonardpitzu, 2019-03-28:
this is default:
`<ons-tab label="Home" icon="ion-home" page="home.html" active></ons-tab>`
key here is the `active` keyword which determines the tab active at startup, i.e. it should look like this:
`<ons-tab label="Map" icon="fa-map-marker" page="map.html" active></ons-tab>`

thundergreen, 2019-03-28:
@leonardpitzu  I guess i will have to build valetudo from source then?

thundergreen, 2019-03-28:
I  but i cannot compile it.

Von: leonardpitzu
Gesendet: Donnerstag, 28. März 2019 15:47
An: Hypfer/Valetudo
Cc: ThUnD3r|Gr33n; Author
Betreff: Re: [Hypfer/Valetudo] A way to show MAP tab on default ? (#127)

this is default:
<ons-tab label="Home" icon="ion-home" page="home.html" active></ons-tab>
key here is the active keyword which determines the tab active at startup, i.e. it should look like this:
<ons-tab label="Map" icon="fa-map-marker" page="map.html" active></ons-tab>

Hypfer, 2019-03-28:
Why don't you just use the map which is available via mqtt?

matthiasharrer, 2019-04-05:
When still using the webinterface this suggestion might be the solution #109

thundergreen, 2019-04-06:
I might have missed that bit how would I get this map via mqtt?

Hypfer, 2019-04-06:
If you enable mqtt the robot sends a buffer with the map as png to `valetudo/rockrobo/map`

At least thats the default topic.

closed 2019-04-12 by thundergreen