Hypfer/Valetudo #131: map.png via mqtt

desq42, 2019-04-01:
Nice Feature to send the map.png via mqtt. (kudos)
I´m not a "Home Assistent"-Guy therefore I´m searching for a way to get the png from my MQTT-Broker which "lives" within ioBroker. My first try to use "node-red" was not successful.
(node-red is unable to decode the binary buffer in which he subscribes to this topic.)
Has somebody been able to subscribe to this topic and "extract" a valid png?
(perhaps via python-script which subscribes and saves the png to disk?)

Regards,
desq

florie1706, 2019-04-01:
Have pretty much the same problem. I use FHEM and it does not show the PNG.it has just funny signs as reading.

Hypfer, 2019-04-02:
[image]
[image]

This looks okay? What was your experience @desq42 ?

desq42, 2019-04-03:
Mine looks like this; (Problem with my mqtt-Broker?)
[image]

Hypfer, 2019-04-03:
Theres definitely something wrong there. PNGs always start with 0x89 0x50 0x4e 0x47. Yours starts with 0xef

There was someone on the Home Assistant discord who also had issues and succeeded with a different mqtt broker so that might indeed be the problem here. It works fine using Mosquitto 1.4.10

desq42, 2019-04-03:
Thanks for the Info. I´m raising an issue for the mqtt-Adapter within ioBroker.
Closing.

closed 2019-04-03 by desq42