Hypfer/Valetudo #133: Provide bin state

Hypfer, 2019-04-02:
Neither the local nor the cloud miio interface seem to provide a way to fetch the current Bin state.
It would be nice if the robot could report when the bin is removed/reinstalled/full/etc.

Since the robot does know about it, the data has to be somewhere. Is it handled by the player process? If yes maybe we could connect to its local socket and fetch the info from there?
Can we get it by sniffing the serial communication between OS and the STM32?
It could also be possible to just log read access onto the relevant wav files. Thats very hackish, though.

cryptomilk, 2019-04-04:
I've also looked into that but haven't had the time to track down what triggers the sound files: bin_in.wav and bin_out.wav

cryptomilk, 2019-04-05:
`/opt/rockrobo/firmware/uart_test can figure it out via -g. It is part of the MCU status, however uart_test can't connect to the serial port, because that's in use by the player software. This means the player is informed about the bin state. Now the question is if we can query the player for the info.

cryptomilk, 2019-04-05:
It looks like the BinCheck is implemented in `/opt/rockrobo/cleaner/bin/RoboController` this opens several pipes in /tmp/

```
lr-x------ 1 root root 64 Apr  5 17:44 10 -> /tmp/NvRc|
lr-x------ 1 root root 64 Apr  5 17:44 11 -> /tmp/ApRc|
l-wx------ 1 root root 64 Apr  5 17:44 12 -> /tmp/RcMc|
lr-x------ 1 root root 64 Apr  5 17:44 13 -> /tmp/McRc|
lr-x------ 1 root root 64 Apr  5 17:44 14 -> /tmp/UpRc|
l-wx------ 1 root root 64 Apr  5 17:44 15 -> /tmp/RcLg|
lr-x------ 1 root root 64 Apr  5 17:44 16 -> /tmp/LgRc|
l-wx------ 1 root root 64 Apr  5 17:44 17 -> /tmp/RcWd|
lr-x------ 1 root root 64 Apr  5 17:44 18 -> /tmp/WdRc|
l-wx------ 1 root root 64 Apr  5 17:44 19 -> /tmp/RcWm|
lr-x------ 1 root root 64 Apr  5 17:44 20 -> /tmp/WmRc|
l-wx------ 1 root root 64 Apr  5 17:44 21 -> /tmp/RcLr|
lr-x------ 1 root root 64 Apr  5 17:44 22 -> /tmp/LrRc|
l-wx------ 1 root root 64 Apr  5 17:44 23 -> /tmp/RcCs|
lr-x------ 1 root root 64 Apr  5 17:44 24 -> /tmp/CsRc|
l-wx------ 1 root root 64 Apr  5 17:44 28 -> /tmp/RcAp|
lr-x------ 1 root root 64 Apr  5 17:44 3 -> /tmp/RcWd|
lr-x------ 1 root root 64 Apr  5 17:44 4 -> /tmp/ApWd|
lr-x------ 1 root root 64 Apr  5 17:44 5 -> /tmp/McWd|
lr-x------ 1 root root 64 Apr  5 17:44 6 -> /tmp/UpWd|
lr-x------ 1 root root 64 Apr  5 17:44 7 -> /tmp/WmWd|
l-wx------ 1 root root 64 Apr  5 17:44 8 -> /tmp/RcNv|
```

I guess we have to figure out how to use them.

cryptomilk, 2019-04-05:
The RoboController connects to the player on port 6665. You can subscribe to events and react on them. Maybe an strace can help to reveal what is sent over that connection.

TheMicroDevil, 2019-04-25:
Would be great to know. My idea was to send the robot to a predefined spot, if dust bin is full, but could not find any information either.

My actual findings:
- Error code 9 => 'Dust bin missing'
- My workaround: Error code 9 followed by error code 0 = dust bin installed
- State 100 => 'Full', maybe this means 'dust bin full', but never saw this state.
- Removing dust bin while device is charging will not throw any error code (/api/current_status) but only trigger the sound files. So state 100 could probably be only seen, when device is not charging.

hagenuck1, 2019-06-12:
I’ve just found out, that there is some kind of counter in /mnt/data/rockrobo/RoboController.cfg

Its „bin_in_time“ and I’ll try to find out in the next days how much it adds to this each clean. 

Would be awesome to have this in the Valetudo UI and published by mqtt to HA, so we could send a notify from HA if the hin could be full. This time it was at 15989, when I cleaned it (looked very full, so cleaning was necessary). 

When I inserted the cleaned bin it sets the value to „0“ within 10 seconds.

cnrd, 2019-06-13:
I am pretty sure the value is cleaning time in seconds since last empty of bin.

Hypfer, 2019-06-13:
https://github.com/Hypfer/Valetudo/issues/17

ADiea, 2019-08-09:
> It looks like the BinCheck is implemented in `/opt/rockrobo/cleaner/bin/RoboController` this opens several pipes in /tmp/
>
> ```
> lr-x------ 1 root root 64 Apr  5 17:44 10 -> /tmp/NvRc|
> lr-x------ 1 [...]

Hi

These pipes seem to be used for inter process communication between RobotController, Player and other processes.
If you use lsof you see exactly what processes have them opened it looks like the naming convention is this: first 2 letters from the Process/Thread for ex /tmp/RcNv would be from RobotController to Navigation thread which is in the player process. 
The communication is binary probably not very hard to crack because we also have some pretty detailed logs for the RobotController

I was interested mainly in the /dev/ttyS2 file which is the uart comm with the MCU

I have spied a little today using strace. I wanted to better understand how everything works. Eventually I would like to make a script that spies on the communication and provides latest state data to a file in /run/shm/ 
Can valetudo pick it up from there if it's in json format and serve it through an api, would that be easy to implement?

888 is the PID of RoboController
```
root@rockrobo:~# lsof -p 888
COMMAND   PID USER   FD   TYPE     DEVICE SIZE/OFF    NODE NAME
RoboContr 888 root  cwd    DIR      179,8     1024   69204 /opt/rockrobo
RoboContr 888 root  rtd    DIR      179,8     1024       2 /
RoboContr 888 root  txt    REG      179,8   244496   69220 /opt/rockrobo/cleaner/bin/RoboController
RoboContr 888 root  mem    REG       0,18   204192    6006 /run/shm/SharedMapInfo
RoboContr 888 root  mem    REG      179,8    34536  100537 /lib/arm-linux-gnueabihf/libnss_files-2.19.so
RoboContr 888 root  mem    REG      179,8    30424  100631 /lib/arm-linux-gnueabihf/libnss_nis-2.19.so
RoboContr 888 root  mem    REG      179,8    55248  100621 /lib/arm-linux-gnueabihf/libnsl-2.19.so
RoboContr 888 root  mem    REG      179,8    22404  100602 /lib/arm-linux-gnueabihf/libnss_compat-2.19.so
RoboContr 888 root  mem    REG       0,18      368    6004 /run/shm/PersistInfo
RoboContr 888 root  DEL    REG        0,4           425997 /SYSV08010000
RoboContr 888 root  mem    REG       0,18     4096    2722 /run/shm/StateChange
RoboContr 888 root  mem    REG      179,8    75400  100446 /lib/arm-linux-gnueabihf/libz.so.1.2.8
RoboContr 888 root  mem    REG      179,8     9824  100628 /lib/arm-linux-gnueabihf/libdl-2.19.so
RoboContr 888 root  mem    REG      179,8   906980  100625 /lib/arm-linux-gnueabihf/libc-2.19.so
RoboContr 888 root  mem    REG      179,8   100012  100464 /lib/arm-linux-gnueabihf/libgcc_s.so.1
RoboContr 888 root  mem    REG      179,8   407136  100612 /lib/arm-linux-gnueabihf/libm-2.19.so
RoboContr 888 root  mem    REG      179,8   723396   68500 /usr/lib/arm-linux-gnueabihf/libstdc++.so.6.0.22
RoboContr 888 root  mem    REG      179,8  2226296   69244 /opt/rockrobo/cleaner/lib/libprotobuf.so.15.0.1
RoboContr 888 root  mem    REG      179,8    55480   69246 /opt/rockrobo/cleaner/lib/librobopersist.so
RoboContr 888 root  mem    REG      179,8     9656   69234 /opt/rockrobo/cleaner/lib/librrlocale.so
RoboContr 888 root  mem    REG      179,8     9780   69248 /opt/rockrobo/cleaner/lib/libpkginfoapi.so
RoboContr 888 root  mem    REG      179,8   119476   69155 /usr/lib/libconfig.so.9.1.3
RoboContr 888 root  mem    REG      179,8   596360   68472 /usr/lib/arm-linux-gnueabihf/libasound.so.2.0.0
RoboContr 888 root  mem    REG      179,8    22580  100632 /lib/arm-linux-gnueabihf/librt-2.19.so
RoboContr 888 root  mem    REG      179,8   102203  100613 /lib/arm-linux-gnueabihf/libpthread-2.19.so
RoboContr 888 root  mem    REG      179,8    44260   69229 /opt/rockrobo/cleaner/lib/libroboipc.so
RoboContr 888 root  mem    REG      179,8    22176   69251 /opt/rockrobo/cleaner/lib/librrlog.so
RoboContr 888 root  mem    REG      179,8   101684  100595 /lib/arm-linux-gnueabihf/ld-2.19.so
RoboContr 888 root  DEL    REG        0,4           851994 /SYSV08020000
RoboContr 888 root  mem    CHR      10,56             1126 /dev/jiffies
RoboContr 888 root  DEL    REG        0,4          5963958 /SYSV07000000
RoboContr 888 root    0r   CHR        1,3      0t0    1030 /dev/null
RoboContr 888 root    1w   REG      179,1   300824      16 /mnt/data/rockrobo/rrlog/watchdog.log
RoboContr 888 root    2w   REG      179,1   300824      16 /mnt/data/rockrobo/rrlog/watchdog.log
RoboContr 888 root    3r  FIFO       0,15      0t0    5462 /tmp/RcWd
RoboContr 888 root    4r  FIFO       0,15      0t0    2646 /tmp/ApWd
RoboContr 888 root    5r  FIFO       0,15      0t0    4396 /tmp/McWd
RoboContr 888 root    6r  FIFO       0,15      0t0    4411 /tmp/UpWd
RoboContr 888 root    7r  FIFO       0,15      0t0    2632 /tmp/WmWd
RoboContr 888 root    8w  FIFO       0,15      0t0    5459 /tmp/RcNv
RoboContr 888 root    9r   REG      179,8       92   69271 /opt/rockrobo/firmware/bin/MCU_FIRMWARE_MATCH
RoboContr 888 root   10r  FIFO       0,15      0t0    4373 /tmp/NvRc
RoboContr 888 root   11r  FIFO       0,15      0t0    4380 /tmp/ApRc
RoboContr 888 root   12w  FIFO       0,15      0t0    2557 /tmp/RcMc
RoboContr 888 root   13r  FIFO       0,15      0t0    4389 /tmp/McRc
RoboContr 888 root   14r  FIFO       0,15      0t0    2604 /tmp/UpRc
RoboContr 888 root   15w  FIFO       0,15      0t0    2558 /tmp/RcLg
RoboContr 888 root   16r  FIFO       0,15      0t0    4412 /tmp/LgRc
RoboContr 888 root   17w  FIFO       0,15      0t0    5462 /tmp/RcWd
[... 35 lines cut ...]
[... cut ...]
(57 is the fd for ttyS2)

```
846   write(57, "\252\22\1\320\2\0\377\300\f\0\0\0\0\0\0\0\0\1\0\0\0\271", 22) = 22
846   write(57, "\252\22\2\320\2\0\1\300\f\0\0\0\0\0\0\0\0\1\0\0\0\251\1", 23) = 23
846   write(57, "\252\22\1\320\2\0\2\300\f\0\0\0\0\0\0\0\0\1\0\0\0003", 22) = 22
```

This being said, using strace (with the cost of some performance drop?) we could spy on the communication of the sensors etc. Probably we will not be able to inject data into the MCU stream, not sure if possible with strace

Hypfer, 2021-01-09:
Use this microservice: https://github.com/johnDorian/rockbin

closed 2021-01-09 by Hypfer