Hypfer/Valetudo #134: Valetudo terminating when publishing MQTT topic

dehrhard, 2019-04-02:
Hello everyone,

I have been trying to get my Roborock 2nd Generation to work with MQTT. I am trying to implement it into FHEM and was successful to the point, that the Device was created. Unfortunately I was not able to get it to execute any command (like "locate").

I confess I am new to the Topic of MQTT and I'm not sure if I understood everything correctly but as far as I can tell, the vaccum should react to commands I publish to the topic "valetudo/rockrobo/command". And I should be able to get a map by publishing to "valetudo/rockrobo/map". When I try to do this though by using MQTT.fx I get no reaction from the Vacuum.

After investigating syslog it shows me that the process is crashing whenever I publish the command:
```
Apr  2 19:13:04 rockrobo kernel: [ 7611.674973] init: valetudo main process (23881) terminated with status 1
Apr  2 19:13:04 rockrobo kernel: [ 7611.675083] init: valetudo main process ended, respawning

```
Is there a way for me to investigate further?
By now I am not sure if this is my Error or if there could be as Issue in the source.
Please correct me if I made any false assumptions. I'm really lost here.

Thanks
Daniel

Hypfer, 2019-04-02:
Please check `/var/log/upstart/valetudo.log` for further information

dehrhard, 2019-04-03:
```
Error: error:06065064:digital envelope routines:EVP_DecryptFinal_ex:bad decrypt
    at Decipheriv.final (crypto.js:183:26)
    at Object.DECRYPT_AND_UNPACK_FILE (/snapshot/Valetudo/lib/Tools.js:498:86)
    at /snapshot/Valetudo/lib/Tools.js:447:35
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3)
Error: error:06065064:digital envelope routines:EVP_DecryptFinal_ex:bad decrypt
    at Decipheriv.final (crypto.js:183:26)
    at Object.DECRYPT_AND_UNPACK_FILE (/snapshot/Valetudo/lib/Tools.js:498:86)
    at /snapshot/Valetudo/lib/Tools.js:447:35
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3)
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: Not supported
    at Parser._parsePayload (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:130:25)
    at Parser.parse (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:37:48)
    at Writable.writable._write (/snapshot/Valetudo/node_modules/mqtt/lib/client.js:301:12)
    at doWrite (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:428:64)
    at writeOrBuffer (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:417:5)
    at Writable.write (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:334:11)
    at Socket.ondata (_stream_readable.js:639:20)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
Loading configuration file: /mnt/data/valetudo/config.json
Webserver running on port 80
```

There are some crypt issues apparently which are thrown regularly but the Parse Error gets thrown when it receives the topic.
It seems to be an issue with the payload?

Hypfer, 2019-04-03:
The crypt issues are most likely related to an unpatched rrlogd + no map in /dev/shm.
https://github.com/mqttjs/mqtt-packet/blob/master/parser.js#L121

🤔
What are you sending?

dehrhard, 2019-04-03:
As I said I'm not too much into MQTT yet so i am shaky on how the payload works. But I was trying to send the "locate" command. I tried it with MQTT.fx and FHEM setList command. Now I am not sure what data type it has to be:

- locate ? 
- "locate" ?
- { locate: "locate"} ?

I tried a few variations but none seemed to work. 

Is there anything I am missing?
Should it be working if I publish "locate" to  valetudo/rockrobo/command ? (rockrobo is the identifier)

Hypfer, 2019-04-03:
Yeah. "locate" should be the correct command. What mqtt broker are you using?

dehrhard, 2019-04-03:
I am using the MQTT2_SERVER Device provided by FHEM. I guess I try an external one?

Hypfer, 2019-04-03:
Yep. Mosquitto should work

dehrhard, 2019-04-04:
I will try that and report back. Thanks

Hypfer, 2019-04-04:
Still, crashing the whole process because a malformed packet arrives shouldn't be happening so this definitely needs error handling

dehrhard, 2019-04-04:
I agree. Still, I wonder what's wrong with the FHEM Broker then. When you add error handling, maybe you could output the corrupted payload.

dehrhard, 2019-04-05:
Small update: I installed mosquitto now and am able to send the "locate" command at long last. So coming back to what you mentioned earlier: So it is not creating the map? It is supposed to publish the map on valetudo/rockrobo/map right? I am not receiving any messages on this topic.

Hypfer, 2019-04-05:
It should start sending map files as soon as you start a cleanup

dehrhard, 2019-04-06:
I was able to access the map now. Thanks a ton!

gsbox, 2019-07-15:
I do have the same issues here. I just updated to 0.4.0 and valetudo only starts with mqtt disabled. Otherwise it gives this error : 
`Error: Not supported
    at Parser._parsePayload (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:130:25)
    at Parser.parse (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:37:48)
    at Writable.writable._write (/snapshot/Valetudo/node_modules/mqtt/lib/client.js:301:12)
    at doWrite (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:428:64)
    at writeOrBuffer (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:417:5)
    at Writable.write (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:334:11)
    at Socket.ondata (_stream_readable.js:663:20)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
Emitted 'error' event at:
    at Parser.emit (events.js:182:13)
    at Parser._emitError (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:373:8)
    at Parser._parsePayload (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:130:14)
    at Parser.parse (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:37:48)
    [... lines matching original stack trace ...]
    at Socket.emit (events.js:182:13)
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Connected successfully to mqtt server
events.js:167
      throw er; // Unhandled 'error' event
      ^`

I hope, this get's solved, because i use FHEM MQTT-Device to send the map via Telegram. 

For now, i just went back to Valetudo 0.3.1 where mqtt is working just fine. 
Thanks for this tremendous piece of software - i love it.

Hypfer, 2019-07-15:
@gsbox

Which MQTT broker are you using?

gsbox, 2019-07-15:
Hi. I am using FHEM-MQTT-Device on my Raspberry Pi. And i would prefer, not to be forced to install another broker, if not really necessary, because my RPi really has a lot to handle with my complete house-installation ;-)

Hypfer, 2019-07-15:
Is there some kind of minimal fhem MQTT setup so I can try to debug this? Like a docker file or something?

gsbox, 2019-07-15:
I do not have a docker environment yet (because a Raspberry Pi it is definetly not the best environment for that ;-) 
But there should be docker images out there with a small fhem-installation. All i did was just the following :
'define fhemMqtt MQTT2_SERVER 9001 global'

which starts a MQTT-Server on Port 9001 on the IP of the Pi. With Valetudo 0.3.1 it works like charm. Are there so many differences regarding MQTT between the two versions ?

Hypfer, 2019-07-15:
Found the culprit.

For whatever reason, the FHEM MQTT Broker sends a mqtt message with type `reserved` which is not handled in the library `mqtt-packet`.

See https://github.com/mqttjs/mqtt-packet/pull/20

Especially this comment
https://github.com/mqttjs/mqtt-packet/pull/20#issuecomment-252989174

which contains this link:
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Table_2.1_-

`Where a flag bit is marked as “Reserved” in Table 2.2 - Flag Bits, it is reserved for future use and MUST be set to the value listed in that table [MQTT-2.2.2-1]. If invalid flags are received, the receiver MUST close the Network Connection [MQTT-2.2.2-2].`

While this throws an error, the connection isn't dropped but apparently just works fine.
So basically to fix this we just need to add an error handler. There should always be one anyways.

This will be fixed in the next release.

closed 2019-07-15 by Hypfer

gsbox, 2019-07-15:
Thanks a lot.

creatvty, 2020-06-13:
Hello, unfortunately this issue is back in the latest release. 
```Connected successfully to mqtt server
events.js:167
      throw er; // Unhandled 'error' event
      ^

Error: Not supported
    at Parser._parsePayload (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:130:25)
    at Parser.parse (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:37:48)
    at Writable.writable._write (/snapshot/Valetudo/node_modules/mqtt/lib/client.js:301:12)
    at doWrite (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:428:64)
    at writeOrBuffer (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:417:5)
    at Writable.write (/snapshot/Valetudo/node_modules/readable-stream/lib/_stream_writable.js:334:11)
    at Socket.ondata (_stream_readable.js:663:20)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
Emitted 'error' event at:
    at Parser.emit (events.js:182:13)
    at Parser._emitError (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:373:8)
    at Parser._parsePayload (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:130:14)
    at Parser.parse (/snapshot/Valetudo/node_modules/mqtt-packet/parser.js:37:48)
    [... lines matching original stack trace ...]
    at Socket.emit (events.js:182:13)
Loading configuration file: /mnt/data/valetudo/config.json
```

Hypfer, 2020-06-13:
@creatvty Which mqtt broker are you using?

creatvty, 2020-06-13:
Also the FHEM Broker

Hypfer, 2020-06-13:
As it turns out, I still have the FHEM docker container including the configuration installed.

I just started it and everything seems to work just fine:
```
[2020-06-13T18:12:32.987Z] [INFO] Connected successfully to mqtt server
[2020-06-13T18:12:32.994Z] [INFO] Connected to non standard compliant MQTT Broker.
```

So you're using Valetudo 0.5.2 with the latest FHEM version? Can you share the relevant FHEM config snippets as well as the used version numbers? @creatvty

creatvty, 2020-06-13:
Sorry my bad. I had 0.5.1 installed. After upgrading to 0.5.2 it worked :disappointed: 
Still a little curious

commit 28149f09 (2019-07-15): Fix #134 crash on FHEM MQTT Broker