Hypfer/Valetudo #136: Logo Design for Valetudo

newfinal100, 2019-04-03:
Hello! I really like this project and I thought I could design a logo for it since it doesn't have one. Here are some variations I made with different shapes of the icon as well as different fonts, you can pick one from these 12 and also let me know if there's anything you'd like me to change.

If you like the idea and the whole look of it, I'll provide you with different color variations so it wont be only black and white, and I'll also send it to you in all formats you may need :) Let me know what you think!

[image]

unfassbarstephan, 2019-04-03:
Great, I really like the idea of a logo. 
 
The designs are very beautiful.  How about combining the V-shaped logo and "aletudo" in a clear font, for example the second one from the upper right square.

But it's only a suggestion.

newfinal100, 2019-04-03:
I'm glad you liked them! And that's a really good idea, I'll try that and send it to you. Also, which V-shaped icon do you like the most from all three? So that I know which one to work with :)

unfassbarstephan, 2019-04-03:
So I think modern clear design is best, so I'm for the V from the 2nd line.  Maybe you can also shorten or straighten the ends a little bit, so that the V comes to the fore even more.  It would also be conceivable to highlight the lettering a little, for example with a dark blue, so that the "Valetudo" is visible at first sight.

newfinal100, 2019-04-03:
[image]

What do you think about these?

unfassbarstephan, 2019-04-04:
I've made some quick and dirty adjustments. The left brighter version looks very promising. I aligned the font according to the orange auxiliary lines and adjusted the size a bit (≈ 120%). 

I like the seconde one in the picture below.

[image]

newfinal100, 2019-04-04:
Done :) is it okay like this?

[image]

Hypfer, 2019-04-04:
Looks great!

newfinal100, 2019-04-04:
Then great :) let me know if there's nothing else you'd like me to change so that I know if I can send you the final original files

Hypfer, 2019-04-04:
I think that it would be a good idea to use the map colors in the logo but thats just a very minor adjustment. Should be easy to change later on.

Other than that I can't say much but thank you. This brings a lot of value to the project and marks another huge step towards the end of beta

newfinal100, 2019-04-04:
I'm very glad that you like it! I agree, it is one huge step :) And yeah, I can definitely do that but I don't know which are the map colors?

Hypfer, 2019-04-04:
See https://github.com/Hypfer/Valetudo/blob/master/client/zone/index.html#L14

unfassbarstephan, 2019-04-04:
I also think the result is great and using the map colors is  good, too.  These are similar shades of blue as well.

Thanks for your work!

newfinal100, 2019-04-04:
[image]

Is this it? :)

Hypfer, 2019-04-04:
Sorry I've pointed to the wrong file since I forgot that I was workin on another branch. It's

#0076ff & #52aeff

newfinal100, 2019-04-04:
It's okay, I'm changing it now :))

newfinal100, 2019-04-04:
[image]

unfassbarstephan, 2019-04-04:
I love it! This is a beautiful color combination.

Hypfer, 2019-04-04:
Perfect!

newfinal100, 2019-04-04:
Glad you liked it! :) here you can get the original files:
https://drive.google.com/drive/folders/1blWNrScNu7zhPN58l3Qj4116ruzZt5uH?usp=sharing

Hypfer, 2019-04-04:
And it's live.

closed 2019-04-04 by Hypfer

axel-kah, 2019-04-04:
Very nice job! A version in square format with just the V part would also be useful as favicon and as "app" shortcut on mobile OS homescreens, etc.

Hypfer, 2019-04-04:
@axel-kah

That's already there. Take a look at the latest commit on master