Hypfer/Valetudo #140: Update documentation for new map upload feature

cryptomilk, 2019-04-05:
I'm opening this so we can discuss what needs to be changed and docs updated.

cryptomilk, 2019-04-05:
I've updated the imagebuilder.sh, could you please check:

https://github.com/cryptomilk/dustcloud/commits/master-wip

cryptomilk, 2019-04-05:
Is the map upload still work in progress or already useable. If it is I could create an image upload and test it. Any steps missing?

Hypfer, 2019-04-05:
Looks good!
That should be everything required to run valetudo 0.3.0

You should be able to build the current cloud_map_upload branch.  ATM only the websocket feature is missing

cryptomilk, 2019-04-05:
If you create a release, don't forget to include everything in the deployment directory, as the script requires it :-)

What is the websocket features, that updates are sent to the browser?

Hypfer, 2019-04-05:
Yeah. Streaming of map updates.

Include into what exactly?

cryptomilk, 2019-04-05:
The new etc/ directory with hosts and rc.local

Hypfer, 2019-04-05:
But it's already in master 🤔

cryptomilk, 2019-04-05:
If you do a release and create a zip file, please include those files :-)

Hypfer, 2019-04-05:
I think it does that automatically.

Anyways: Everything seems to be finished and working. There are of course some TODOs remaining, but I think it's in a stable enough state to publish a release.
Now I just need to complete the Installation guide in the Valetudo wiki + write some release notes.

cryptomilk, 2019-04-05:
With the map upload feature, you can cleanup `/mnt/data/` now?

`rm -rf /mnt/data/rockrobo/rrlog/000*` ???

cryptomilk, 2019-04-05:
Hmm, it created a new directory in `/mnt/data/rockrobo/rrlog`.

I didn't turn on writing rrlog. Hmm.

cryptomilk, 2019-04-05:
I think I've found it. Pushed changes for imagebuilder.sh

Hypfer, 2019-04-05:
When will you open the PR?

cryptomilk, 2019-04-06:
I can do and tell Dennis to only merge it once Valetudo 0.3.0 is released.

cryptomilk, 2019-04-06:
The FAQ on the website says valetudo is the name of hygenia. Searching for it on the web reveals:

> Vale tudo
> Vale tudo is an unarmed, full-contact combat sport with relatively few rules.

I think that would also fit the robots job description :-)

Hypfer, 2019-04-06:
😄
Dennis can merge ASAP. I got everything ready and just need to push it to github

oechslein, 2019-04-06:
On https://github.com/dgiese/dustcloud/wiki/Cloud-Free-Firmware-Image-With-Valetudo it is mentioned to install dummycloud:
```
wget https://github.com/dgiese/dustcloud/releases/download/0.1/dummycloud_0.1.zip
unzip -a dummycloud*.zip
```
Since this is no longer needed we do have to remove it, or?

Hypfer, 2019-04-06:
@oechslein there's a new guide in the wiki of this repo which unfortunately cannot be used yet since the PR https://github.com/dgiese/dustcloud/pull/207 hasn't been merged yet since its currently 05:56 in Boston.

The guide can be found here: https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions
It applies to Valetudo 0.3.0 which hasn't been released yet due to the above.

I assume that in a few hours this should all be done

closed 2019-04-06 by Hypfer