Hypfer/Valetudo #143: No map with latest cloud_map_upload branch

cryptomilk, 2019-04-06:
I've just compiled the latest code from the cloud_map_upload branch. I don't have a map anymore. I just see the robot icon.

I'm running 69c264aa232f3b869901e4d0f46bf1be1086a902

Valetudo should really write a log file.

Hypfer, 2019-04-06:
Did you install it via scp? Maybe a reboot is necessary.
This happens when a map is requested but the robot hasn't uploaded any, yet. Its a default 1x1 Pixel map so that nothing breaks which results in a giant robot.

If everything is configured correctly, a map should appear after around 60 seconds since thats the update interval.
Still, I'd recommend rebooting if you haven't done that already

Hypfer, 2019-04-06:
If your lab mode is set to 0 and/or there is no persistent map available, it might be possible that you first need to do a cleanup so that a map appears 🤔

cryptomilk, 2019-04-06:
```
root@rockrobo:~# cat /mnt/data/rockrobo/lab.cfg 
1
```

Hypfer, 2019-04-06:
If you're seeing `Robot connected` in `/var/log/upstart/valetudo.log` it should be functional. Just start a cleanup then.

Still it would be nice to have more information on that available: https://github.com/Hypfer/Valetudo/issues/145

cryptomilk, 2019-04-06:
After a reboot the map is back. Does Valetudo only store the map in memory?

Hypfer, 2019-04-06:
Yep since I figured that the robot already stores it and we can just ask for it anytime

cryptomilk, 2019-04-06:
But why doesn't it work after a service restart?

I've started a cleanup and it even diddn't get it.

Hypfer, 2019-04-06:
Did you stop dummycloud, replaced the iptables rules and then scp'd valetudo?

I don't know exactly whats happening there but I also experienced that. Rebooting did solve it though.
@matthiasharrer did experience this as well

I guess it's temporary. Maybe network related?

cryptomilk, 2019-04-06:
Yesterday I've built a new FW just with valetudo. It worked just fine. This morning I fetched your latest changes, rebuilt valetudo and copied the new binary to the robot, started valetudo and only got the robot icon in the map.

Hypfer, 2019-04-06:
iirc, the robot did respond with "retry" on each map upload request in this scenario, so the cloud connection is there but the miio_client is unable to upload the map to the local server.

cryptomilk, 2019-04-06:
Could you add more logging, like when we request a new map, when the map was updated. Like a log line when it changes a state so it is easier to understand from the log what is going on?

cryptomilk, 2019-04-06:
Btw. ware you on Freenode or Matrix?

Hypfer, 2019-04-06:
I just registered #valetudo on freenode :-)

closed 2019-04-07 by Hypfer